solettaproject/soletta #1376: make samples failed with ./src/samples/flow/oic/light-scan/light-client-scan.fbp

yongli3, 2016-02-01:
make samples failed with ./src/samples/flow/oic/light-scan/light-client-scan.fbp

"make samples V=1" failed with the below error message:
     GEN   build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c
mkdir -p ./build/stage/samples/flow/oic/light-scan/
./build/soletta_sysroot/usr/bin/sol-fbp-generator -j ./build/soletta_sysroot/usr/share//soletta//flow//descriptions/ ./src/samples/flow/oic/light-scan/light-client-scan.fbp ./build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c  -c ./src/samples/flow/oic/light-scan/sol-flow.json
WRN:sol-fbp ./src/bin/sol-fbp-generator/main.c:752 get_node_data() nd->desc == NULL
ERR:sol-fbp ./src/bin/sol-fbp-generator/main.c:2482 create_fbp_data() Failed to resolve node type.
tools/build/Makefile.rules:384: recipe for target 'build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c' failed
make: *** [build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c] Error 1

otaviobp, 2016-02-01:
I couldn't reproduce that on top of Soletta master branch. Building using "make alldefconfig && make samples" is working fine.
Can you give me more information about the configuration used to build and which branch you are using.
Thanks

yongli3, 2016-02-02:
Thanks @otaviobp I am using the latest master branch:
git branch -v
* master 9aff6c0 comms:dtls: Fix pointer signedness issue

make distclean; make alldefconfig; make samples V=1 reported the below build error message:
mkdir -p ./build/stage/samples/flow/oic/light-scan/
./build/soletta_sysroot/usr/bin/sol-fbp-generator -j ./build/soletta_sysroot/usr/share//soletta//flow//descriptions/ ./src/samples/flow/oic/light-scan/light-client-scan.fbp ./build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c  -c ./src/samples/flow/oic/light-scan/sol-flow.json
WRN:sol-fbp ./src/bin/sol-fbp-generator/main.c:752 get_node_data() nd->desc == NULL
ERR:sol-fbp ./src/bin/sol-fbp-generator/main.c:2482 create_fbp_data() Failed to resolve node type.
tools/build/Makefile.rules:384: recipe for target 'build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c' failed
make: *** [build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c] Error 1

The build system is Ubuntu 1504 64bit. 

There is another build error message if I remove these files in src/thirdparty/tinycbor/
     GEN   build/stage/samples/flow/power-supply/battery-capacity-gen.c
WRN:sol-fbp ./src/bin/sol-fbp-generator/main.c:752 get_node_data() nd->desc == NULL
ERR:sol-fbp ./src/bin/sol-fbp-generator/main.c:2482 create_fbp_data() Failed to resolve node type.
tools/build/Makefile.rules:384: recipe for target 'build/stage/samples/flow/power-supply/battery-capacity-gen.c' failed
make: *** [build/stage/samples/flow/power-supply/battery-capacity-gen.c] Error 1

otaviobp, 2016-02-02:
I still couldn't reproduce the build. Can you send me the .config used in the build and the config.log file?

Thanks

otaviobp, 2016-02-02:
I suspect that you may have a problem in any submodule. Can you try to do the following:
git submodule init && git submodule sync && git submodule update
Then:
make distclean && make alldefconfig && make samples V=1

bdilly, 2016-02-02:
Yong Li's log:

root@intel-quark:~# ls -l /usr/share/soletta/flow/descriptions/oic.json 
-rw-r--r-- 1 root root 4639 Jan 28 14:50 /usr/share/soletta/flow/descriptions/oic.json
root@intel-quark:~# ls -l /usr/lib/soletta/modules/flow/oic.so 
-rwxr-xr-x 1 root root 31528 Jan 28 14:50 /usr/lib/soletta/modules/flow/oic.so
root@intel-quark:~# export SOL_MACHINE_ID="580a3d6a9d194a23b90a24573558d2f4"
root@intel-quark:~# export SOL_LOG_LEVEL=5
root@intel-quark:~# sol-fbp-runner light-server.fbp
DBG:sol-platform ./src/lib/common/sol-platform.c:138 sol_platform_get_board_name() detected board name=intel-galileo-rev-g
DBG:sol-platform ./src/lib/common/sol-platform.c:161 sol_platform_get_board_name() using board name=intel-galileo-rev-g
INF: ./src/lib/common/sol-pin-mux.c:100 _load_mux() Loaded pin multiplexer 'intel-galileo-rev-g' from '//usr/lib/soletta/modules//pin-mux//intel-galileo-rev-g.so'
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:144 sol_init() Soletta 0.0.1 on linux-4.1.15-yocto-standard initialized
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:183 sol_run() run
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:114 sol_flow_resolve() could not resolve module for id='oic/server-light' using resolver=builtins_resolver
DBG:sol-flow ./src/lib/flow/sol-flow-resolver-conffile.c:210 _resolver_conffile_get_module() module named 'oic' loaded from '//usr/lib/soletta/modules//flow//oic.so'
WRN:sol-flow ./src/lib/flow/sol-flow-resolver-conffile.c:213 _resolver_conffile_get_module() Type='oic/server-light' not found.
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:114 sol_flow_resolve() could not resolve module for id='oic/server-light' using resolver=conffile
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:114 sol_flow_resolve() could not resolve module for id='oic/server-light' using resolver=parse-state-resolver-with-declares
light-server.fbp:37:1 Couldn't resolve type 'oic/server-light' of node 'light'
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:207 sol_quit_with_code() quit with code 1
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:224 sol_shutdown() shutdown
root@intel-quark:~#

yongli3, 2016-02-03:
Thanks @otaviobp I followed your steps, but still get the build error, below is the output message:
git branch -v
* master a158685 Build: Set -ffloat-store on non 64 bits architetures.

git submodule init && git submodule sync && git submodule update
Synchronizing submodule url for 'src/thirdparty/duktape'
Synchronizing submodule url for 'src/thirdparty/mavlink'
Synchronizing submodule url for 'src/thirdparty/tinycbor'
Synchronizing submodule url for 'src/thirdparty/tinydtls'

make distclean && make alldefconfig && make samples V=1

mkdir -p ./build/stage/samples/flow/oic/light-scan/
./build/soletta_sysroot/usr/bin/sol-fbp-generator -j ./build/soletta_sysroot/usr/share//soletta//flow//descriptions/ ./src/samples/flow/oic/light-scan/light-client-scan.fbp ./build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c  -c ./src/samples/flow/oic/light-scan/sol-flow.json
WRN:sol-fbp ./src/bin/sol-fbp-generator/main.c:752 get_node_data() nd->desc == NULL
ERR:sol-fbp ./src/bin/sol-fbp-generator/main.c:2482 create_fbp_data() Failed to resolve node type.
tools/build/Makefile.rules:384: recipe for target 'build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c' failed
make: *** [build/stage/samples/flow/oic/light-scan/light-client-scan-gen.c] Error 1

I uploaded the .config and config.log file as .txt files

[- Copy.config.txt](https://github.com/solettaproject/soletta/files/114997/-.Copy.config.txt)
[config.log.txt](https://github.com/solettaproject/soletta/files/114998/config.log.txt)

otaviobp, 2016-02-03:
I found the problem you are facing. Config and config.log files helped a lot. Thanks.
To be able to build light-client-scan sample your machine needs to have gtk installed and gtk node type should be selected. Anyway, this wasn't suppose to break the build system. There was some dependencies missing in some samples and that was causing the break.
The bug was fixed and a pull request was created at: #1388
Building Soletta using the build from the pull request should fix your problem.

Thanks,

otaviobp, 2016-02-03:
Pullreq #1388 was merged to master branch. Can you try to build Soletta master branch and check if your problem was fixed?
Thanks

yongli3, 2016-02-04:
@otaviobp  After git pull to the latest 96b80d8, make samples okay now

closed 2016-02-04 by yongli3