solettaproject/soletta #1380: OIC Server: Interoperability issues with iotivity

gabrielschulhof, 2016-02-01:
Two sequences produce unexpected results.

OS: Fedora 21 32-bit with kernel 4.1.13-100.fc21.i686
iotivty-1.0.1
soletta@7a15fa7bf74b9e2681da640354a72e499411170b

Sequence 1:
1. Run ```ocserverbasicops```
2. Run ```oic-sample-server```
At this point oic-sample-server will complain that the address is already in use.

Sequence 2:
1. Run ```oic-sample-server```
2. Run ```ocserverbasicops```
3. Run ```occlient -t 1```
At this point the client discovers three copies of ```/sol/0``` provided by ```oic-sample-server```, but no instances of ```/a/led``` provided by ```ocserverbasicops```

barbieri, 2016-02-01:
Is this unexpected?

AFAIU, if you use OIC you listen in a specific UDP port. Then if that is on a single machine, you can't run multiple instances.

The sequence 1 shows that soletta is restrictive and doesn't allow to share the UDP port (ie: no `SO_REUSEADDR`), while the sequence 2 shows that iotivity allows reusing the port... but then you're in trouble since only one process is getting the UDP packets, likely just soletta.

This is similar to running multiple HTTP servers all at port 80.

It is expected that you run each server on its own machine, maybe a virtual machine for tests.

In environments where people expect multiple applications to do OIC we need another infrastructure to be a proxy. @thiagomacieira may know the details of how OIC wants to deal with that.

gabrielschulhof, 2016-02-01:
Well, as it stands, iotivity picks a random available port. For example, the first instance of ocserverbasicops gets port 37414 whereas the second instance of ocserverbasicops gets port 58898. So, you can keep running any number of servers on the same machine as long as you have free ports.

This makes end-to-end testing with multiple servers much, much easier than having to fire up a VM for each server.

barbieri, 2016-02-01:
doesn't make much sense to me, so likely we're missing some detail that iotivity may use underneath, like one automatically proxying to the other.

Do you know what happens at iotivity side, @thiagomacieira ?

gabrielschulhof, 2016-02-01:
iotivity's ```OCInit()```, which initializes the stack, takes a port number which may be 0, in which case it auto-selects an available port. ```sol_coap_server_new()``` allows the same thing. I use it client-side as ```sol_coap_server_new(0);```, but server-side I don't get to initialize the coap server itself, because the server-side is simply ```int sol_oic_server_init (void)```, so, presumably, the coap server is stored in a global variable somewhere.

thiagomacieira, 2016-02-02:
IoTivity opens 8 sockets:

I. For IPv6:
   a. Secured (DTLS):
      1. A well-known port to receive multicast (5684)
      2. A random port  to receive unicast
   b. Not encrypted
      3. A well-known port to receive multicast (5683)
      4. A random port to receive unicast
II. For IPv4: same as above

All the well-known ports need to be opened with SO_REUSEADDR and SO_REUSEPORT (rant about why there are two suppressed). Because of that, it should never send from one of those ports (with one exception)

Not all of the ports make sense, but IoTivity opens them anyway. You can reduce the number of open sockets from the theoretical maximum:

* DTLS over multicast doesn't exist and OIC hasn't implemented secure discovery
* no production device should be communicating via unencrypted unicast, so sockets 4 & 8 are not necessary
* One single IPv6 socket with IPV6_V6ONLY turned off is enough for unicast on all platforms that implement the standard IPv6 socket API, so sockets 2&6 and 4&8 can be the same
* IPv4 is optional on devices not operating on Ethernet or WiFi (sockets 5-8)
* On Linux, you can do both IPv4 and IPv6 multicasting on one IPv6 socket (pairs 1/5 and 3/7)
* If you know that your OIC stack is the only stack requiring CoAP on this device, you can use the same socket for unicast and multicast (pairs 1/2, 3/4, 5/6, 7/8)

So the theoretical minimum is one socket bound to 5684 and one on 5683. A typical scenario would be 3 sockets:
1. IPv6 port 5683 (IPV6_V6ONLY on)
2. IPv4 port 5683
3. Random port on both IPv4 and IPv6 (IPV6_V6ONLY off)

thiagomacieira, 2016-02-02:
Note an important detail about running multiple stacks on the same device: the sender must configure the socket for IPV6_MULTICAST_LOOP /  IP_MULTICAST_LOOP, which means the sender will receive its own discovery queries.

IoTivity does not yet implement this.

barbieri, 2016-02-02:
Great insights. 

But I still think that reusing the same port for 2 processes will lead only one to receive the data gram in these "well known" ports. 

@lpereira and @otaviobp can you double check our implementation matches what Thiago said?

thiagomacieira, 2016-02-02:
Hence the SO_REUSEADDR and SO_REUSEPORT, plus the multicast loop.

ceolin, 2016-02-02:
as far as i could see our implementation does not set these properties in the socket.

thiagomacieira, 2016-02-02:
Remember that if you set multicast loop, you need to drop your looped back queries.

ceolin, 2016-02-02:
i'll keep it in mind. Btw, iotivity is not setting SO_REUSEPORT, are you aware of this ?

thiagomacieira, 2016-02-02:
It's using SO_REUSEADDR. Rant about why there are two is suppressed.

ceolin, 2016-02-02:
i afraid i've not understood you. These options are different right ? And for well known ports it should being using SO_REUSEPORT instead of SO_REUSEADDR.
Is it right or i am missing something ?

thiagomacieira, 2016-02-02:
They're different and have different behaviours. SO_REUSEPORT is new on Linux. And people don't understand when to use one and when to use the other, or both. I read the kernel source code and I **still** don't understand.

So the answer is "use the one that is correct", where you'll have to find out what "correct" is.

lpereira, 2016-02-02:
SO_REUSEADDR is useful specially when testing: it takes some time for a bound socket to be cleaned up, and starting the server again after a crash or after a fresh compilation might require some waiting. This sort of bypasses that timeout (at least in the TCP case, I'm not really familiar with UDP).

SO_REUSEPORT is of limited use. Two use cases: starting a new version of a server without closing current connections, and load balancing when you don't have threading support (or when you have something akin to Python's GIL) to fully exploit many CPUs/cores. In the first case, when the old server is about to be reloaded, this flag must be set in the bound socket, accept() shouldn't be called in it anymore, and then the new server starts up normally; when all connections in the old server are dealt with, it can close itself -- the kernel will deliver new connections only to the new process. In the second case, many copies of a server process can be opened, and the kernel will pick which process blocked on accept() (or poll()) to wake up and tell the good news.

thiagomacieira, 2016-02-02:
Which one should you use when you want two applications to bind to the same UDP port?

Generally, on all interfaces, but also consider that some may want to bind specifically to some IP addresses and not all.

lpereira, 2016-02-02:
That would be SO_REUSEPORT, but then the kernel would only send datagrams to one process (either at random, or round robin, or in some other way), potentially to the wrong one. I can't see this working for the intended use case.

thiagomacieira, 2016-02-02:
Then SO_REUSEADDR it is.

Quick test, open 2 x python and paste
```python
import socket
f=socket.socket(socket.AF_INET6, socket.SOCK_DGRAM, 0)
f.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, 1)
f.bind(('::', 1234))
f.setsockopt(socket.IPPROTO_IPV6, socket.IPV6_JOIN_GROUP, b'\xff\x03\0\0\0\0\0\0\0\0\0\0\0\0\0\xfd\0\0\0\0')
f.recvfrom(1)
```

In a third shell, run:
  nc -w0 -u ff03::1 1234 <<<''

Make sure your firewalls allow incoming UDP packets to port 1234.

Result: both instances get a UDP datagram of 1 byte.

Test 2: replace SO_REUSEPORT with SO_REUSEADDR. Same result.

lpereira, 2016-02-03:
So the handling of these options is different in TCP and UDP. Pretty curious. I can see now why you were puzzled trying to figure out their differences even after reading the kernel code.

thiagomacieira, 2016-02-03:
I remember from the source code that it differs on what happens if the one of the sockets is bound to an IP and the other is bound to 0.0.0.0 or ::. This is called "port hijacking" and I've seen Coverity complain about it.

I just can't explain it. Hence the rant.

thiagomacieira, 2016-02-03:
As noted on iotivity-dev: IPV6_MULTICAST_LOOP is enabled by default.

barbieri, 2016-02-03:
likely the behavior is different for UDP and TCP because with UDP the datagram is fixed and can be easily reference counted, so you get the reference until you deliver to all users. For TCP it's a moving window, some process can consume 5 bytes, while the other consumes 1k... so cumbersome to deliver to all.

thiagomacieira, 2016-02-03:
It doesn't make sense for TCP, as it needs to be ACK'ed and it's supposed to be reliable. If two sockets were receiving the same data and one of the applications froze, the kernel would have to stop ACK'ing once its buffer got full. The other application, not frozen, would stop receiving data.

UDP is not reliable. If the kernel buffers start to fill up, the kernel can decide to drop packets left and right.

ceolin, 2016-03-04:
This issue pointed is fixed on soletta upstream. Soletta is following the iotivity behaviour/implementation.

closed 2016-03-04 by ceolin