solettaproject/soletta #1386: Magnet IIO node reading hang

yongli3, 2016-02-03:
Reading the light and gyro nodes are okay, but the magnet iio node failed, there is no error just reading hang, below is the test fbp file and log message, how to trace down this issue?

magnet(iio/magnet:iio_device="hmc5843 create,i2c,platform/80860F41:05,0x1e,hmc5843 3",buffer_size=-1)
timer(timer)

root@intel-corei7-64:/# /magnet.fbp
INF:sol-platform ./src/lib/common/sol-platform-linux-common.c:961 sol_platform_impl_load_locales() The locale file (/etc/locale.conf) was not found in the system.
INF: ./src/shared/sol-util-file.c:423 sol_util_iterate_dir() Could not open dir [/etc//soletta/boards/] to iterate: No such file or directory
DBG:sol-platform ./src/lib/common/sol-platform.c:218 sol_platform_get_board_name() board name is unknown
DBG:sol-update ./src/lib/common/sol-update.c:119 get_first_module_on_dir() Invalid update module dir: //usr/lib/soletta/modules//update/
DBG:sol-update ./src/lib/common/sol-update.c:151 load_update_module() No update module to load
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:187 sol_init() Soletta 0.0.1 on linux-4.1.15-yocto-standard initialized
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:229 sol_run() run
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:114 sol_flow_resolve() could not resolve module for id='iio/magnet' using resolver=builtins_resolver
DBG:sol-flow ./src/lib/flow/sol-flow-resolver-conffile.c:211 _resolver_conffile_get_module() module named 'iio' loaded from '//usr/lib/soletta/modules//flow//iio.so'
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='iio/magnet' resolved using resolver=conffile
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='iio/magnet' resolved using resolver=parse-state-resolver-with-declares
DBG:sol-flow ./src/lib/flow/sol-flow-node-options.c:773 fill_options_with_named_options() Parsed option "iio_device" member #0 name="iio_device", type="string", offset=8, size=8
DBG:sol-flow ./src/lib/flow/sol-flow-node-options.c:773 fill_options_with_named_options() Parsed option "buffer_size" member #1 name="buffer_size", type="int", offset=16, size=4
DBG:sol-flow ./src/lib/flow/sol-flow-builder.c:519 sol_flow_builder_add_node() Node gyro added: type=0x7f4f891ede00, opts=0x61ae00.
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='timer' resolved using resolver=builtins_resolver
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='timer' resolved using resolver=parse-state-resolver-with-declares
DBG:sol-flow ./src/lib/flow/sol-flow-builder.c:519 sol_flow_builder_add_node() Node timer added: type=0x7f4f8c8776c0, opts=0x619fa0.
DBG:sol-flow ./src/lib/flow/sol-flow-builder.c:950 sol_flow_builder_get_node_type() Node type 0x61af90 created
INF: ./src/shared/sol-util-file.c:423 sol_util_iterate_dir() Could not open dir [/etc//soletta/boards/] to iterate: No such file or directory
DBG:sol-platform ./src/lib/common/sol-platform.c:218 sol_platform_get_board_name() board name is unknown
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow-magnet.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: //sol-flow-magnet.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: //sol-flow.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: /etc//soletta//sol-flow-magnet.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: /etc//soletta//sol-flow.json
WRN:sol-flow-iio ./src/modules/flow/iio/nodes.c:285 magnet_open() magnet_open
DBG:sol-flow ./src/lib/flow/sol-flow-static.c:242 connect_nodes() Making 0 connections.
DBG: ./src/lib/io/sol-iio.c:1401 create_or_resolve_device_address_dispatch() IIO device creation/resolving dispatching command: hmc5843
DBG: ./src/lib/io/sol-iio.c:712 sol_iio_open() iio device created. device3 - buffer_enabled: 0 - manual_trigger: 0 - trigger_name: (null) - trigger_id: 0
INF: ./src/lib/io/sol-iio.c:854 channel_get_offset() Could not get offset for channel [in_magn_x] in device3. Assuming 0
DBG: ./src/lib/io/sol-iio.c:978 sol_iio_add_channel() channel [in_magn_x] added. scale: 0.000008 - offset: 0 - storagebits: 0 - bits: 0 - mask: 0
INF: ./src/lib/io/sol-iio.c:854 channel_get_offset() Could not get offset for channel [in_magn_y] in device3. Assuming 0
DBG: ./src/lib/io/sol-iio.c:978 sol_iio_add_channel() channel [in_magn_y] added. scale: 0.000008 - offset: 0 - storagebits: 0 - bits: 0 - mask: 0
INF: ./src/lib/io/sol-iio.c:854 channel_get_offset() Could not get offset for channel [in_magn_z] in device3. Assuming 0
DBG: ./src/lib/io/sol-iio.c:978 sol_iio_add_channel() channel [in_magn_z] added. scale: 0.000008 - offset: 0 - storagebits: 0 - bits: 0 - mask: 0
WRN:sol-flow-iio ./src/modules/flow/iio/nodes.c:269 magnet_create_device_cb() magnet_create_device_cb return!

edersondisouza, 2016-02-03:
> Reading the light and gyro nodes are okay, but the magnet iio node
> failed, there is no error just reading hang, below is the test fbp file
> and log message, how to trace down this issue? [...]

Hi,

I guess there's something missing on the fbp code you've written here. 
There's no connection between 'timer' and 'magnet' nodes. Could you 
please send the whole fbp code you are using? Again, 'strace -ff' can be 
of great help in this case, too.

Cheers,

yongli3, 2016-02-03:
@edersondisouza  Thanks for your comment, below is the fbp file for your reference, I am using almost the same fbp format to test other sensors. I will capture the "strace -ff" log later
 cat magnet.fbp
#!/usr/bin/env sol-fbp-runner
# This file is part of the Soletta Project
#
# Copyright (C) 2015 Intel Corporation. All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions
# are met:
#
#   * Redistributions of source code must retain the above copyright
#     notice, this list of conditions and the following disclaimer.
#   * Redistributions in binary form must reproduce the above copyright
#     notice, this list of conditions and the following disclaimer in
#     the documentation and/or other materials provided with the
#     distribution.
#   * Neither the name of Intel Corporation nor the names of its
#     contributors may be used to endorse or promote products derived
#     from this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
# "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
# LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
# A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
# OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
# SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
# LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
# DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
# THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
# (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
# OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

# Print iio gyroscope reading each second
# iio_device: will first try any device whose name is 'l3g4200d'. If fails,
# will try to create it using the 'create' command. If it fails, tries
# IIO device whose id is '0'.

magnet(iio/magnet:iio_device="hmc5843 create,i2c,platform/80860F41:05,0x1e,hmc5843 3",buffer_size=-1)
timer(timer)

edersondisouza, 2016-02-03:
> @edersondisouza <https://github.com/edersondisouza> Thanks for your
> comment, below is the fbp file for your reference, I am using almost the
> same fbp format to test other sensors. I will capture the "strace -ff"
> log later
> cat magnet.fbp
> #!/usr/bin/env sol-fbp-runner

[...]

> magnet(iio/magnet:iio_device="hmc5843
> create,i2c,platform/80860F41:05,0x1e,hmc5843 3",buffer_size=-1)
> timer(timer)

Well, then it's indeed missing a connection between 'timer' and 'magnet' 
nodes.
You may want to add something like

     timer OUT -> TICK magnet OUT -> IN _(console)

To the end of the fbp file. Then it shall print magnet readings each second.

Cheers!

yongli3, 2016-02-03:
THANKS @edersondisouza! issue fixed. It is a copy-paste typo, I forgot the last line.

closed 2016-02-03 by yongli3

edersondisouza, 2016-02-03:
> THANKS @edersondisouza <https://github.com/edersondisouza>! issue fixed.
> It is a copy-paste typo, I forgot the last line.

=D