solettaproject/soletta #1396: sol-network & sol-socket: move strings to sol-buffer

barbieri, 2016-02-03:
Both network and socket operations that do manipulate strings should be dealing with buffers (`struct sol_buffer`) instead of getting `char *buf` and `size_t` (in some functions) or doing `strdup()` in others.

Keep a single API that receives a buffer, grow the buffer as needed and give to the function (some functions allow to query the size) or call `sol_buffer_append()`.

Remember to document if buffers must be initialized before or not. I prefer to have people to provide initialized buffers since they can put one that points to the stack and says "fixed_capacity" so no allocations are done.

It would be nice to review the whole API to to provide a single behavior across soletta (all functions outputting strings to buffers).

@cabelitos and @ceolin should know how to do this

cabelitos, 2016-02-19:
@barbieri Hello,
Don't you think that this change will make the API more difficult to use? Imagine the case that I wan't to receive a message from the socket.
```C
void handle_msg(sol_socket *socket, struct sol_network_link_addr cliaddr)
{
    uint8_t mem[DTLS_MAX_BUF];
    struct sol_buffer buf = SOL_BUFFER_INIT_FLAGS(mem, sizeof(mem),
        SOL_BUFFER_FLAGS_FIXED_CAPACITY);
    int len;

  len = sol_socket_recvmsg(socket, &buf, &cliaddr);

  //Do my stuff here
}
```

ibriano, 2016-02-19:
On Fri, 19 Feb 2016 12:57:46 -0800, Guilherme Íscaro wrote:
> @barbieri Hello,
> Don't you think that this change will make the API more difficult to use? Imagine the case that I wan't to receive a message from the socket.
> ```C
> void handle_msg(sol_socket [...]

But that's not a string.

cabelitos, 2016-02-19:
@ibriano Indeed, however I assumed that char * meant bytes, not purely string.

cabelitos, 2016-02-19:
Anyway, ignore my comments. Thank you

cabelitos, 2016-02-23:
#1461

cabelitos, 2016-02-27:
merged

closed 2016-02-27 by cabelitos