solettaproject/soletta #1403: I/O APIs: add UART write with tx buffer as byte array

zolkis, 2016-02-05:
Currently the signature for UART send is using ```unsigned char *tx``` for transmit buffer:
https://github.com/solettaproject/soletta/blob/master/src/lib/io/include/sol-uart.h#L150
```javascript
bool sol_uart_write(
    struct sol_uart *uart,
    const unsigned char *tx, unsigned int length,
    void (*tx_cb)(void *data, struct sol_uart *uart, unsigned char *tx, int status),
    const void *data);
```
Shouldn't the transmit buffer be a byte array ~~instead~~ in addition?
For instance other APIs like SPI use ```const uint8_t *tx``` which is would be more correct IMO also for UART, but you could also keep the string interface for convenience and backward comp.
https://github.com/solettaproject/soletta/blob/master/src/lib/io/include/sol-spi.h#L118

The reason for asking is that we are having a problem mapping the UART JS API which currently accepts a byte array or a string.
https://github.com/solettaproject/soletta/blob/master/doc/js-spec/uart.md
If the clients provides a byte array, we need to internally convert to a string because Soletta uses a string. In the other way it would make more sense, and having both APIs would make even more sense - therefore suggesting adding one more signature for write.

Just asking - what was the rationale for this signature? Did you have most use cases with clients sending text through UART rather than binary data? What are the typical expected uses cases (sensors) that use UART?

zolkis, 2016-02-05:
If ```unsigned char *``` is actually defined as ```uint8_t *``` on all platforms then of course forget the above. Anyway using ```uint8_t *``` would make it more clear.

barbieri, 2016-02-09:
I guess @ibriano forgot to change these, it should be `uint8_t *`

ibriano, 2016-02-12:
Done

closed 2016-02-12 by ibriano