solettaproject/soletta #1407: OIC: Client fails to discover resource created with soletta server

gabrielschulhof, 2016-02-09:
https://gist.github.com/gabrielschulhof/235d19629bcaaec82c1f

Seen with f04a302c8b10c9ec7eb6937ddf482804b35f0f2e

Steps to reproduce:

0. Make sure no OIC servers are running on your network
1. In one terminal: start ```server```
2. In another terminal: start ```client```

At this point, ```client``` hangs and ultimately reports ```null resource```, i.e. discovery timeout.

To compare:

0. Make sure no OIC servers are running on your network
1. In one terminal: start ```ocserverbasicops``` from iotivity 1.0.1
2. In another terminal: start ```client```

At this point, ```client``` returns almost immediately with ```/oic/d```.

barbieri, 2016-02-09:
btw, these kind of client/server.c should be in our src/samples/oic, if not there yet.

gabrielschulhof, 2016-02-11:
Hmmm ... I guess without the active flag, the resource is not visible even if it is discoverable.

closed 2016-02-11 by gabrielschulhof