solettaproject/soletta #1413: OIC C API: server not able to define path/href

poussa, 2016-02-12:
The OIC server C API call `sol_oic_server_add_resource(sol_oic_resource_type *, ...)` is used to define the resources the OIC server has. However, the API does not allow the app/user to define what the path (or href) of the resource is. Instead, the implementation auto generates the path (`/sol/0`, `/sol/1`, etc).  

The OIC server should be able to define what the path of it's resources are, like `/a/light`, `/room/door`. The current API does not allow this to be possible.

otaviobp, 2016-02-12:
We have a pull request adding an API for that. take a look at #1410

poussa, 2016-02-12:
The PR seems to address this case. Once landed we can close this.

closed 2016-02-16 by otaviobp