solettaproject/soletta #1442: Auto load the iio-trig-sysfs module

yongli3, 2016-02-19:
If the iio-trig-sysfs kernel module is not loaded, the below error message occurs:
INF: ./src/lib/io/sol-iio.c:344 check_trigger() No current trigger for iio:device0. Creating a sysfs one.
WRN: ./src/lib/io/sol-iio.c:290 create_sysfs_trigger() No 'iio_sysfs_trigger' under '/sys/bus/iio/devices'. Missing 'modprobe iio-trig-sysfs'?

It should be better if the create_sysfs_trigger function can auto load the module

barbieri, 2016-02-19:
we link to kmod, so we can add that feature... however it will not be that useful in practice since usually applications won't be executed as root and `modprobe` needs such functionality.

Maybe it's better to just fix this in the OS by adding it to `/etc/modules-load.d/iio-trig-sysfs.conf`?

yongli3, 2016-02-19:
@barbieri Thanks for your comment, I noticed there is a file low-speed-spidev.conf in the /etc/modprobe.d, maybe we can use the same solution?

barbieri, 2016-02-19:
exactly