solettaproject/soletta #1469: use of python3

bapclenet, 2016-02-24:
```
The following (required) dependencies were not met:

python module: jsonschema

If you've just installed it, run: make reconf
For more information/options, run: make help
```

You should add that you use python 3 here beacause I tried to install jsonschema with pip (which was linked to python 2.7) and Soletta couldn't find it after the install because it has not been added to python3

ibriano, 2016-02-24:
Ah, those are listed as dependencies somewhere in Soletta's documentation.

dorileo, 2016-02-24:
Since dependency-resolver.py is a python3 script this will consider all the python tests/checks to be python3, the message can be changed to reflect it. I'm cooking a simple patch for that.

dorileo, 2016-02-24:
pull request #1470 fixes the issue.

bapclenet, 2016-02-24:
Ok thanks

closed 2016-02-24 by bapclenet