solettaproject/soletta #1487: Platform: Removing/adding monitor from monitor callback results in infinite loop

gabrielschulhof, 2016-02-26:
#### Issue Description
When you remove a platform monitor from within the callback for a platform monitor and then re-add it again, you end up with an infinite loop because the monitor is re-added at the end of the queue of monitors to be called and is not marked as do-not-call-this-time-around.

Basically, when iterating over the list of attached monitors, one needs to account for the fact that the attached monitors may alter the list itself. As a simple heuristic, monitors added while the list of monitors is being walked should not be executed until the next event.

#### Steps to reproduce

 1. Build and run the attached code
 2. Change the hostname

At this point the program enters an infinite loop.

#### Board information

Host OS and architecture: Fedora 23 64-bit

Target OS and platform: Fedora 23 64-bit

#### Logs

```C
#include <stdio.h>
#include <sol-platform.h>
#include <sol-mainloop.h>

static void hostnameChanged( void *data, const char *newName ) {
	printf( "%s\n", newName );
	sol_platform_del_hostname_monitor( hostnameChanged, NULL );
	sol_platform_add_hostname_monitor( hostnameChanged, NULL );
}

static void startup() {
	sol_platform_add_hostname_monitor( hostnameChanged, NULL );
}

static void shutdown() {
}

SOL_MAIN_DEFAULT(startup, shutdown);
```

cabelitos, 2016-02-26:
This is lovely

cabelitos, 2016-02-27:
This is lovely.

barbieri, 2016-02-28:
weird, this uses the `sol_monitors` that we wrote to handle that exact case you describe.

In `sol-platform.c`, function `sol_platform_inform_hostname_monitors()` will call your `hostnameChanged()` using `SOL_MONITORS_WALK()`, this macro will increment `walking` counter that is later checked by `sol_monitors_append()` which is called by `monitor_add()/monitor_add_and_register()`. 

**EXCEPT** the `sol_monitors_append()`  is not checking `walking` :-/ only `sol_monitors_del()` does.

This is pretty much the problem we had to handle in the main-loop, we shouldn't dispatch new elements during that loop. To do so is easy: change the `SOL_MONITORS_WALK()` to cache a local copy of `ms->entries.len` and limit the loop to that amount of elements.

What is left to do is investigate `SOL_MONITORS_WALK()` usages to see if any requires the current approach, I really don't think there is such thing. If there is not, then we fix the single macro and get all fixes for free.

Thanks @gabrielschulhof for reporting.

@cabelitos can you take a look at it?

cabelitos, 2016-02-29:
@barbieri I thought the same thing when I read the bug description. I will take a look and provide a fix for it.

cabelitos, 2016-02-29:
#1500

barbieri, 2016-02-29:
#1504 looks good, just minor details, please fix and commit directly.

cabelitos, 2016-03-01:
Fixed.

closed 2016-03-01 by cabelitos