solettaproject/soletta #1516: tools/run-fbp-tests script raise exception because logging instantiation is incorrect

otcyanglei, 2016-03-02:
#### Task Description
run fbp test with --log option, like:
$ tools/run-fbp-tests --runner "/usr/bin/sol-fbp-runner" --log "DEBUG" "src/test-fbp/"

#### Dependencies
n/a

#### Issue Description

The runner tools/run-fbp-tests raise exception and stops when run with --log option:

Traceback (most recent call last):
  File "run-fbp-tests", line 354, in <module>
    set_log_level(args)
  File "run-fbp-tests", line 325, in set_log_level
    hdlr = root.handlers[0]
IndexError: list index out of range

#### Steps to reproduce

 1. $ $ tools/run-fbp-tests --runner "/usr/bin/sol-fbp-runner" --log "DEBUG" "src/test-fbp/"
 
#### Board information
n/a

Host OS and architecture:
Ubuntu 14.04 / python 2.7

Target OS and platform:
n/a

#### Logs

n/a
test script tool issue.

otcyanglei, 2016-03-02:
According to https://docs.python.org/2/library/logging.html#module-level-functions，the logging instantiation in tools/run-fbp-runner is incorrect:
324     root = logging.getLogger()
325     hdlr = root.handlers[0]
326     fmt = logging.Formatter('%(message)s')
327     hdlr.setFormatter(fmt)
328     hdlr.setLevel(level)

before invoke getLogger(), python script should call basicConfig() to ensure that at least one handler is available.

otcyanglei, 2016-03-02:
Submit PR #1517 to fix this issue.

bdilly, 2016-03-03:
PR #1517 was integrated, so I'm closing this issue.
Thanks

closed 2016-03-03 by bdilly