solettaproject/soletta #1521: flow: missing alignment/padding for forwarded options

barbieri, 2016-03-03:
When we use a FBP that contains `OPTION=node.option:external_option` the type sizes may result in incorrect ordering. Example:

```
INPORT=state.IN:IN_STATE
INPORT=power.IN:IN_POWER

OUTPORT=state.OUT:OUT_STATE
OUTPORT=power.OUT:OUT_POWER

OPTION=state.default_value:default_state # boolean, size = 1
OPTION=power.default_value:default_power # int32_t, size = 4, needs 4 byte alignment, placed at offset "1"

state(persistence/boolean:name="state",storage=fs)
power(persistence/int:name="power",storage=fs)
```

Error:
```
src/lib/flow/sol-flow-node-options.c:651:25: runtime error: store to misaligned address 0x07a142ae for type 'int32_t', which requires 4 byte alignme
nt
0x07a142ae: note: pointer points here
==14085== Syscall param write(buf) points to uninitialised byte(s)
==14085==    at 0x7145357: syscall (in /usr/lib/libc-2.22.so)
==14085==    by 0x6BBFEE6: __sanitizer::internal_write(int, void const*, unsigned long) (sanitizer_linux.cc:148)
==14085==    by 0x6BC31A3: __sanitizer::IsAccessibleMemoryRange(unsigned long, unsigned long) (sanitizer_posix_libcdep.cc:175)
==14085==    by 0x6BB2D35: renderMemorySnippet (ubsan_diag.cc:242)
==14085==    by 0x6BB2D35: __ubsan::Diag::~Diag() (ubsan_diag.cc:332)
==14085==    by 0x6BB5264: handleTypeMismatchImpl(__ubsan::TypeMismatchData*, unsigned long, __ubsan::Location, __ubsan::ReportOptions) (ubsan_handl
ers.cc:62)
==14085==    by 0x6BB5F45: __ubsan_handle_type_mismatch (ubsan_handlers.cc:68)
==14085==    by 0x499F111: set_member (sol-flow-node-options.c:651)
==14085==    by 0x499F92F: fill_options_with_named_options (sol-flow-node-options.c:764)
==14085==    by 0x49A0631: sol_flow_node_options_new (sol-flow-node-options.c:861)
==14085==    by 0x49A92DD: build_node (sol-flow-parser.c:463)
==14085==    by 0x49AA474: build_flow (sol-flow-parser.c:848)
==14085==    by 0x49AB28A: sol_flow_parse_buffer (sol-flow-parser.c:967)
==14085==  Address 0x7a142ad is 5 bytes inside a block of size 10 alloc'd
==14085==    at 0x482A1AE: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==14085==    by 0x49A02E0: sol_flow_node_options_new (sol-flow-node-options.c:841)
==14085==    by 0x49A92DD: build_node (sol-flow-parser.c:463)
==14085==    by 0x49AA474: build_flow (sol-flow-parser.c:848)
==14085==    by 0x49AB28A: sol_flow_parse_buffer (sol-flow-parser.c:967)
==14085==    by 0x10DE25: runner_new_from_file (runner.c:416)
==14085==    by 0x10B2A9: startup (main.c:199)
==14085==    by 0x4916EB0: sol_mainloop_common_idler_process (sol-mainloop-common.c:268)
==14085==    by 0x491AC10: sol_mainloop_impl_iter (sol-mainloop-impl-posix.c:744)
==14085==    by 0x4918045: sol_mainloop_impl_run (sol-mainloop-common.c:553)
==14085==    by 0x491B574: sol_run (sol-mainloop.c:231)
==14085==    by 0x10BC59: main (main.c:271)
==14085==
```

cabelitos, 2016-03-03:
I will take a look

cabelitos, 2016-03-04:
#1534

cabelitos, 2016-03-08:
Fixed

closed 2016-03-08 by cabelitos