solettaproject/soletta #1564: fbp: allow to omit ports if only one exists

barbieri, 2016-03-07:
While manually writing FBP it is inconvenient to type the port name if the node provides a single port, namely this:

```sh
_(gpio/reader) OUT -> IN _(console)
```

could be expressed more concisely as:
```sh
_(gpio/reader) -> _(console)
```

otaviobp, 2016-03-08:
+1 for this issue too.
But I have some comments about it:
 - Nodes have error output ports, so we should be implicit for nodes that a single port, except error ports.
 - If another port is added to the node, we break the node API. And that is not happening with current version of fbp.

I suggest to make OUT and IN ports implicit. So independently of the number of ports a node have, we use IN/OUT ports if no port is explicitly selected.

ibriano, 2016-03-08:
On Tue, 08 Mar 2016 05:09:59 -0800, Otávio wrote:
> +1 for this issue too.
> But I have some comments about it:
>  - Nodes have error output ports, so we should be implicit for nodes that a single port, except error ports.
>  - If another port is added to [...]

And what to they map to? And what type do they have?

barbieri, 2016-03-14:
he means if out output port is specified, you assumed "OUT", if there is a port with such name. It's an "alias".

bdilly, 2016-03-29:
I don't like that too much, actually

Imo explicit is better than implicit. Also you're creating one extra way to write the same thing.
I don't think it makes it much easier to read.

Port names should be descriptive and help users to understand what should be feed to nodes and what should be the outcomes.

glima, 2016-04-28:
-1 for this as well.