solettaproject/soletta #1565: fbp: easier constants

barbieri, 2016-03-07:
We could translate constants to node-types with option "value" being the actual contents. The node-type to be used depends on the connecting port. Examples:

```sh
'abc' -> IN _(console) # defaults to string if input port is any
'true' -> IN _(gpio/writer)
```

Would be automatically converted to:
```sh
_(constant/string:value="abc") OUT -> IN _(console)
_(constant/boolean:value="true") OUT -> IN _(gpio/writer)
```

@glima you did a previous attempt at initial packets before, would you see a problem with this approach? it just come to me recently and would not require changes to the core, no new methods or "from string" as we tried before -- it was my personal issue with that. With this we'd need to change the parser to do some conversion and deeper look at the structure to do the replacement with the actual node.

@cmarcelo ^

otaviobp, 2016-03-08:
That would be really a good improvement.

cmarcelo, 2016-03-08:
> 'abc' -> IN _(console) # defaults to string if input port is any
> 'true' -> IN _(gpio/writer)

I think we never disagreed on this idea at FBP level. So looks good to me.

> With this we'd need to change the parser to do some conversion

Change code in `sol-flow-parser.c` and above it. I would still represent these as separate entities at the syntax level (the code in files named fbp).

> no new methods or "from string" as we tried before

The node will do a conversion, so "from string" is there, just the way to reach it is different. Will this feature be available for non-builtin packets? If so, how to select which node to create for each packet type?

One idea would be to have some sort of convention: given a packet type xyz, if a certain node type exists (and that had appropriate characteristics), it would be used. Another option is starting without supporting this for custom packet types.

barbieri, 2016-03-14:
I'd not do that for non-builtin, looks like a good compromise. Buit-in would have to be more explicit, unless we find a way to extend the concept.