solettaproject/soletta #1583: Samples:Grove Kit is lacking information

gpaes, 2016-03-10:
#### Task Description
The Samples:Grove Kit page (https://github.com/solettaproject/soletta/wiki/Samples:Grove-Kit) is lacking information about the samples:
- It would be nice to have an introduction about what the samples are about
- There must be a link to the source code/flow of each sample
- There must be an explanation about what each sample do
- There must be an explanation about the pins, the person might not use the pictured board to test the sample

#### Dependencies

mbelluzzo, 2016-03-18:
@gpaes, I kind of understood what you mean by "explanation about the pins", but independently of having a picture or not, because the pins don't need to be that exactly ones and that many pins have the needed capability, It would be insane to list everywhere: "you could use any pin from x to y", specially when there are holes. It also changes from board to board and we won't have pictures for every board.

barbieri, 2016-03-18:
We must describe based on our example, not the possibilities. You can say why it was chosen and how user could find an alternative.

mbelluzzo, 2016-03-18:
But that is already different in each config file

gpaes, 2016-04-05:
@mbelluzzo Can't we do an explanation based on @barbieri approach? That's what I was aiming for with that task.

mbelluzzo, 2016-04-05:
As I answered him, the pins are already different between config files of the same sample for different boards. I think is going a little too far on silly details, inside the config itself it says type: pwm for example, pin 5, why? it has pwm support, obvious. type gpio, pin: 10, why, has gpio, what is the alternative? any other pin that makes you happy that supports gpio. Is that simple.

gpaes, 2016-04-05:
@mbelluzzo Yes, they are different for each config file. Is it too much to explain "this is connected to a pwm pin" / "this is connected to gpio" and so on? Won't this be a nice information to have stated along the example?

mbelluzzo, 2016-04-05:
To explain that is a pwm on a pwm pin is implied on the sample explanation. To explain the alternatives and why a given pin was chosen (not pin type, iow pin 3 instead of pwm pin) as it was suggested, not.

edersondisouza, 2016-04-05:
> To explain that is a pwm on a pwm pin is implied on the sample
> explanation. To explain the alternatives and why a given pin was chosen
> (not pin type, iow pin 3 instead of pwm pin) as it was suggested, not.

Maybe all that is necessary is to explain something like:

"Here you see that sensors are connected to the same pins used on fbp 
samples. You may want to choose them, to any compatible pin - but 
remember to change fbp code accordingly"

?

mbelluzzo, 2016-04-05:
+2 on only add this disclaimer after sample explanation.

gpaes, 2016-04-05:
@edersondisouza @mbelluzzo That makes sense:)

closed 2016-04-06 by mbelluzzo