solettaproject/soletta #1586: sol-http-server.h support keep connection open after response was sent.

barbieri, 2016-03-10:
In order to implement #1585 and #1582 we need a way to send a response with code + headers and then allow information to be sent progressively.

One way is to steal connection's file description and allow direct `write(2)` calls to it. In this case clients would be responsible to monitor if data could be written, add `sol_fd_add()` to monitor non-blocking calls, etc. It's simpler and may be enough for these cases. Another option is to append the desired memory to a buffer, register the `sol_fd_add()` internally until the buffer is made empty. It is more convenient but may grow memory usage in places where outside could keep the buffer themselves. In the issues linked above, they would need to do the buffer on their side, they don't keep that themselves (or they could be lazy and turn the socket blocking and block on `write(2)` calls, not that great). But for large data, like reading from a file or doing transcoding from media elements, then it could stop the feeding pipeline and not blow server's memory usage.

Assigning to @ceolin since he wrote microhttpd's server implementation, but feel free to reassign to others.

ceolin, 2016-03-18:
https://github.com/solettaproject/soletta/pull/1665

ceolin, 2016-03-21:
https://github.com/solettaproject/soletta/pull/1682

ceolin, 2016-03-31:
integrated

closed 2016-03-31 by ceolin