solettaproject/soletta #1601: Analog AIO interface not working

sragavan, 2016-03-14:
#### Task Description
Analog AIO api doesn't seem to be working. Tested with the below steps on flow as well.
#### Dependencies

#### Issue Description

#### Steps to reproduce

Open the device ip:8080 flow editor and open the Groove sample for Temperature sensor.
Run the source and it reports -273.150000C. Which is 0K and it should report real temperature.

#### Board information

Intel Edison/Soletta from github

mbelluzzo, 2016-03-14:
If you are using Soletta Dev-app, the patch that fix this issue is'nt in yet (commit '38073d'). It should be in for beta 16. What you can do to fix this issue in the meanwhile is to compile Soletta master branch and overwrite the installation on your dev-app image.

glima, 2016-03-30:
@mbelluzzo: is this fixed now?

mbelluzzo, 2016-03-30:
yep

closed 2016-03-30 by mbelluzzo