solettaproject/soletta #1602: OIC Server: Observation not working

gabrielschulhof, 2016-03-14:
#### Non-polling observation gives COM_ERROR on the iotivity-node side

When attempting to observe the server created with https://gist.github.com/gabrielschulhof/3f750a0065366c0a417b#file-observable-server-c, one gets OC_STATUS_COM_ERROR in response to an observe request.

#### Steps to reproduce

0. Build libsoletta.so
1. Build https://gist.github.com/gabrielschulhof/3f750a0065366c0a417b#file-observable-server-c from the soletta root with ```cc -g -Wall -O0 -I$(pwd)/build/soletta_sysroot/usr/include/soletta -L$(pwd)/build/soletta_sysroot/usr/lib -lsoletta -Wl,-rpath $(pwd)/build/soletta_sysroot/usr/lib ../observable-server.c -o ./observable-server``
2. Run ```./observable-server```
3. Build ```iotivity-node```
4. Edit ```js/client.observe.js``` from ```iotivity-node``` and set ```sampleUri = "/a/something"```
5. Run ```node js/client.observe.js``` from iotivity-node

Expected: The iotivity-node client finds the server and outputs the result of the observe request whenever the server issues a notification.

Actual: The observe request handler returns with ```OC_STACK_COMM_ERROR(29)```.

#### Board information

Host OS and architecture: Fedora 23 64-bit

Target OS and platform: Fedora 23 64-bit

#### Logs

 * Soletta version (top commit hash): ece42bf713a7396ea603be37c992526257a6384c

otaviobp, 2016-03-17:
This issue was introduced by commit: ef354016: coap: Allows two servers at same time.
If you revert that, it is suppose to work fine. Can you test that, please?

@ceolin Have you found a solution to the problems related to this change? I vote for reverting this and having a single instance of server running in the same machine if we have no solution for this problem.

ceolin, 2016-03-29:
The problem is we are using the CoAP port for unicast packets. We should open and use a random port for the unicast packets and use the CoAP port for the multicast.
Reverting that commit we do not solve the problem since only one process will receive the data which is even worse.

barbieri, 2016-05-10:
use the P3 label instead of using the subject.

What's the status on this? @otaviobp, @gabrielschulhof ?

otaviobp, 2016-05-11:
This issue was caused to the fix implemented for this task: https://github.com/solettaproject/soletta/issues/1380
We need to have a socket listening in a random port for unicast packets and the CoAP port for multicast, as @ceolin said.
Note that this issue is only occurring because we have a port clash that is only happening when running oic server and oic client in the same machine, so this won't happen in real use scenarios.

ibriano, 2016-05-11:
On Wed, 11 May 2016 06:30:37 -0700, Otávio wrote:
> This issue was caused to the fix implemented for this task: https://github.com/solettaproject/soletta/issues/1380
> We need to have a socket listening in a random port for unicast packets and the CoAP [...]

Why? Why does the client need a specific port? Running it along side a
server shouldn't be a problem.

otaviobp, 2016-05-11:
It looks that IoTivity is creating a server socket even if we are creating an OIC client. In samples they use that for /oic/ad resource.
I just tested the reported problem and it is not happening when SO_REUSEADDR is not used.

bdilly, 2016-05-11:
@barbieri we use the subject because QA guys were reporting issues and weren't able to set labels since they don't have write access to the repo. Back at that time there weren't any options to enable it for them.

Sure, since they don't do it anymore, we could use labels anyway.

otaviobp, 2016-06-20:
Fixed by #2190

otaviobp, 2016-06-21:
fixed

closed 2016-06-21 by otaviobp