solettaproject/soletta #1622: MinnowBoard MAX Instructions: Improve page text

gpaes, 2016-03-15:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/MinnowBoard-MAX-Instructions
Review text on "Image generation", add brief paragraphs for the steps.
Review text on "Boot" and its list of steps. While obvious to some it's nice to state what to expect after the final step is taken.

#### Dependencies
None

bdilly, 2016-07-19:
Looks like somebody already added description for image generation steps.

I've added a couple of links regarding the boot sections, where users may find more information if they need - no reason to duplicate this info.

closed 2016-07-19 by bdilly

gpaes, 2016-07-21:
Just for the record, the address of the page changed to: https://github.com/solettaproject/soletta/wiki/Building-Soletta%E2%84%A2-Framework-for-MinnowBoard-MAX