solettaproject/soletta #1628: iio node fails to initialize in time before called

yma11, 2016-03-16:
#### Issue Description
Would like to use node constant/empty to tick sensor read gyro() twice. But it fails since iio node haven't finished initialized before the TICK

#### Steps to reproduce

 1. refer sample from https://github.com/solettaproject/soletta/wiki/Logic-constructs-on-Soletta-flows and compose gyroscope sensor data read code as following:
#!/usr/bin/env sol-fbp-runner
tick_1(constant/empty)
tick_2(constant/empty)
gyro(iio/gyroscope:iio_device="mpu6050 create,i2c,platform/80860F41:05,0x68,mpu6050 0",buffer_size=20)
tick_1 OUT -> TICK gyro
tick_2 OUT -> TICK gyro

 2. run it using command: export SOL_LOG_LEVEL='DBG' && ./test_mpu6050_tick_3.fbp

#### Board information

Host OS and architecture:

Target OS and platform:
Ostro/MinnowMax
#### Logs
Since pastebin temporarily unavailable from my side, I put log here:

INF:sol-platform ./src/lib/common/sol-platform-linux-common.c:960 sol_platform_impl_load_locales() The locale file (/etc/locale.conf) was not found in the system.
INF: ./src/shared/sol-util-file.c:419 sol_util_iterate_dir() Could not open dir [/etc//soletta/boards/] to iterate: No such file or directory
DBG:sol-platform ./src/lib/common/sol-platform.c:193 sol_platform_get_board_name() detected board name=intel-minnow-max-linux_gt_3_17
DBG:sol-platform ./src/lib/common/sol-platform.c:216 sol_platform_get_board_name() using board name=intel-minnow-max-linux_gt_3_17
INF:sol-pin-mux ./src/lib/common/sol-pin-mux.c:82 _load_mux() Could not load platform pin multiplexer '//usr/lib/soletta/modules//pin-mux//intel-minnow-max-linux_gt_3_17.so': //usr/lib/soletta/modules//pin-mux//intel-minnow-max-linux_gt_3_17.so: cannot open shared object file: No such file or directory
DBG:sol-update ./src/lib/common/sol-update.c:120 get_first_module_on_dir() Invalid update module dir: //usr/lib/soletta/modules//update/
DBG:sol-update ./src/lib/common/sol-update.c:152 load_update_module() No update module to load
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:187 sol_init() Soletta 0.0.1 on linux-4.1.15-yocto-standard initialized
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:229 sol_run() run
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='constant/empty' resolved using resolver=builtins_resolver
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='constant/empty' resolved using resolver=parse-state-resolver-with-declares
DBG:sol-flow ./src/lib/flow/sol-flow-builder.c:519 sol_flow_builder_add_node() Node tick_1 added: type=0x7fcd883570a0, opts=0x62a940.
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='constant/empty' resolved using resolver=builtins_resolver
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='constant/empty' resolved using resolver=parse-state-resolver-with-declares
DBG:sol-flow ./src/lib/flow/sol-flow-builder.c:519 sol_flow_builder_add_node() Node tick_2 added: type=0x7fcd883570a0, opts=0x62ae60.
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:114 sol_flow_resolve() could not resolve module for id='iio/gyroscope' using resolver=builtins_resolver
DBG:sol-flow ./src/lib/flow/sol-flow-resolver-conffile.c:210 _resolver_conffile_get_module() module named 'iio' loaded from '//usr/lib/soletta/modules//flow//iio.so'
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='iio/gyroscope' resolved using resolver=conffile
DBG:sol-flow ./src/lib/flow/sol-flow-resolver.c:118 sol_flow_resolve() module for id='iio/gyroscope' resolved using resolver=parse-state-resolver-with-declares
DBG:sol-flow ./src/lib/flow/sol-flow-node-options.c:773 fill_options_with_named_options() Parsed option "iio_device" member #0 name="iio_device", type="string", offset=8, size=8
DBG:sol-flow ./src/lib/flow/sol-flow-node-options.c:773 fill_options_with_named_options() Parsed option "buffer_size" member #1 name="buffer_size", type="int", offset=16, size=4
DBG:sol-flow ./src/lib/flow/sol-flow-builder.c:519 sol_flow_builder_add_node() Node gyro added: type=0x7fcd8281b7a0, opts=0x658df0.
DBG:sol-flow ./src/lib/flow/sol-flow-builder.c:950 sol_flow_builder_get_node_type() Node type 0x658ff0 created
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow-test_mpu6050_tick_3-intel-minnow-max-linux_gt_3_17.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow-test_mpu6050_tick_3.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow-intel-minnow-max-linux_gt_3_17.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow-test_mpu6050_tick_3-intel-minnow-max-linux_gt_3_17.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow-test_mpu6050_tick_3.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow-intel-minnow-max-linux_gt_3_17.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: ./sol-flow.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: /etc//soletta//sol-flow-test_mpu6050_tick_3-intel-minnow-max-linux_gt_3_17.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: /etc//soletta//sol-flow-test_mpu6050_tick_3.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: /etc//soletta//sol-flow-intel-minnow-max-linux_gt_3_17.json
DBG: ./src/shared/sol-conffile.c:724 _load_json_from_paths() Trying to load conffile: /etc//soletta//sol-flow.json
DBG:sol-flow ./src/lib/flow/sol-flow-static.c:242 connect_nodes() Making 2 connections.
WRN: ./src/lib/io/sol-iio.c:1090 sol_iio_device_trigger_now() device == NULL
WRN:sol-flow-iio ./src/modules/flow/iio/nodes.c:192 gyroscope_tick() Could not read channel values
WRN: ./src/lib/io/sol-iio.c:1090 sol_iio_device_trigger_now() device == NULL
WRN:sol-flow-iio ./src/modules/flow/iio/nodes.c:192 gyroscope_tick() Could not read channel values
DBG: ./src/lib/io/sol-iio.c:1385 create_or_resolve_device_address_dispatch() IIO device creation/resolving dispatching command: mpu6050
WRN: ./src/lib/io/sol-iio.c:357 set_buffer_size() Could not set IIO device buffer size
DBG: ./src/lib/io/sol-iio.c:710 sol_iio_open() iio device created. device0 - buffer_enabled: 1 - manual_trigger: 1 - trigger_name: mpu6050-dev0 - trigger_id: 0
INF: ./src/lib/io/sol-iio.c:852 channel_get_offset() Could not get offset for channel [in_anglvel_x] in device0. Assuming 0
DBG: ./src/lib/io/sol-iio.c:969 sol_iio_add_channel() channel [in_anglvel_x] added. scale: 0.001065 - offset: 0 - storagebits: 16 - bits: 16 - mask: 65535
INF: ./src/lib/io/sol-iio.c:852 channel_get_offset() Could not get offset for channel [in_anglvel_y] in device0. Assuming 0
DBG: ./src/lib/io/sol-iio.c:969 sol_iio_add_channel() channel [in_anglvel_y] added. scale: 0.001065 - offset: 0 - storagebits: 16 - bits: 16 - mask: 65535
INF: ./src/lib/io/sol-iio.c:852 channel_get_offset() Could not get offset for channel [in_anglvel_z] in device0. Assuming 0
DBG: ./src/lib/io/sol-iio.c:969 sol_iio_add_channel() channel [in_anglvel_z] added. scale: 0.001065 - offset: 0 - storagebits: 16 - bits: 16 - mask: 65535
WRN:sol-flow ./src/lib/flow/sol-flow-static.c:337 flow_send_do() Error packet '5 (Could not read channel values)' sent from './test_mpu6050_tick_3.fbp (0x6590e0)' was not handled
WRN:sol-flow ./src/lib/flow/sol-flow-static.c:337 flow_send_do() Error packet '5 (Could not read channel values)' sent from './test_mpu6050_tick_3.fbp (0x6590e0)' was not handled

edersondisouza, 2016-03-23:
Hi @yma11,

Please, check if #1715 fix the issues. Note however that I added a note to sample you refer to: it may not always work. Some sensors need a time between two consecutive readings.

glima, 2016-03-30:
@yma11: have you had the time to check that for us? We will assume it's fixed soon and you'll have to re-open if that's not the case. Thanks.

yma11, 2016-04-07:
@glima: sorry to reply you late but can you tell me how can I verify the fix works when this PR in open status? can I get a soletta image which includes this fix? thanks.

glima, 2016-05-10:
@yma11, yes: https://github.com/solettaproject/soletta/releases/tag/v1_beta19

edersondisouza, 2016-05-11:
@yma11 I'll close this issue. If you happen to find it again, feel free to reopen it.

closed 2016-05-11 by edersondisouza