solettaproject/soletta #1630: Segmentation fault on sol-fbp-runner

cristina-moraru, 2016-03-16:
#### Issue Description

The tool sol-fbp-runner gets Segmentation Fault at malformed input

#### Steps to reproduce

 1. Build Soletta locally from branch master with last commit hash: 774c37cd2302006b773b75a2e4c5face9dd5f506.

Config file http://pastebin.com/HDLM4jaF

 2. Run 
sol-fbp-runner input_file 

The input file contains non-ASCII characters so I am attaching a hexdump of the file:
http://pastebin.com/FAAP6GfX

 3.

#### Board information

Same Host and Target
Parameters:
uname -m = x86_64
uname -r = 3.16.0-48-generic
uname -s = Linux
uname -v = #64~14.04.1-Ubuntu SMP Thu Aug 20 23:03:57 UTC 2015

#### Logs

Backtrace of the crash:
http://pastebin.com/vt8Uy2Sy

The problem arises from an illegal memory access at *ptr in file src/modules/flow/form/form.c:292

barbieri, 2016-03-21:
Well, the version you ran is outdated, please try it with a more recent version as your test does not crash my git-master. Also, for all kinds of crashes or invalid memory operations attach the valgrind, see https://github.com/solettaproject/soletta/wiki/How-to-report-an-issue

However, looking at the code written by @glima, it is not correctly handling the invalid formats when tags are missing. The code says "Bad format... Fallbacking to pristine", then set the tag to a newly allocated value using `sol_util_replace_str_if_changed()`. Here is the first problem as this memory is not released. Then the `format_chunk()` proceeds by using `ptr` and comparing it to `end_ptr`, but `ptr` is the value tag, which was allocated and thus is not in the segment between `end_ptr`.

The "correct path", as I understand, is to keep a reference to `{title}` and `{value}` inside the given format strings (`title` or `format` options). If the option is empty or does not contain the replacement, then it will fail.

Then I'm, re-assigning this to @glima, he wrote the code and is easier to review that.

glima, 2016-03-21:
Ouch, this version of yours is too old, @cristina-moraru! @bdilly reported some fuzzy crash on form to me a while ago and I did a commit for that that seems to have addressed the issue. We had even a new beta release after that (beta16). So please test with at least the new release, I also could not see anything crash here. @barbieri, all the new tests I put in were run with valgrind with 0 leaks or bad accesses. I also fail to see were a bad comparison is being made, since the ptrs will always point to the same allocated memory (either the one strduped or the one replaced_if_changed()).

brunobottazzini, 2016-03-29:
@cristina-moraru, have you got the change to test the sol-fbp-runned updated ?

glima, 2016-03-29:
Closing, as it seems to have been resolved. Please reopen if it's not the case.

closed 2016-03-29 by glima

cristina-moraru, 2016-03-30:
Indeed I was using an outdated tree but I updated it an run the tests again. This time no crash was found.