solettaproject/soletta #1631: Malformed json is accepted by sol-fbp-generator

cristina-moraru, 2016-03-16:
#### Issue Description

Certain malformed json files pass the checks and the tool creates malformed output.c files. The tool checks the json structure and the values of 'name' keys but all other values are put into the output.c file even if they are invalid.

#### Steps to reproduce

 1. Build Soletta locally from branch master with last commit hash: 774c37c.
     Config file http://pastebin.com/HDLM4jaF

 2. Run 
sol-fbp-generator -j malformed-timer.json -j console.json -j boolean.json simple.fbp malformed-output.c
 Where files can be found at:

The malformed-timer.json contains non-ascii characters so I am attaching a hexdump:
http://pastebin.com/HFhDPARg

The rest of the files are part of soletta tree
./src/modules/flow/console/console.json
./src/modules/flow/boolean/boolean.json
./src/samples/flow/basics/simple.fbp

#### Board information

Same Host and Target
Parameters:
uname -m = x86_64
uname -r = 3.16.0-48-generic
uname -s = Linux
uname -v = #64~14.04.1-Ubuntu SMP Thu Aug 20 23:03:57 UTC 2015

#### Logs

The output file malformed-output contains the invalid values from malformed-timer.json:
A hexdump of the malformed-output.c:  http://pastebin.com/Ktv8701e

barbieri, 2016-03-21:
ok, so to summarize this problem: we're not validating the `_symbol` fields, we just pass it as is to the generated file. This is because we provide a JSON schema that should be used to validate it. We could add a runtime verification when we load it, shouldn't impact us much and will help users to understand their problem.

However this is not a bug due the schema not being used as designed.

cabelitos, 2016-03-23:
#1712

cabelitos, 2016-03-24:
@cristina-moraru I merged a fix for this problem. Could you please test it again?
commit e2f150c

cristina-moraru, 2016-03-28:
I tested again and now it seems to be fine.

brunobottazzini, 2016-03-29:
Closed.

Since the issue was solved

Thanks

closed 2016-03-29 by brunobottazzini