solettaproject/soletta #1632: Edison Instructions:

gpaes, 2016-03-16:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/Edison-Instructions
Improve page text to make it easier to understand what's the content of this document and navigate through it.

- [ ] Add an introduction text to explain what these instructions are for and briefly talk about what are the steps going to be taken.
- [ ] Review/reword Yocto introduction text ("To build it first it is needed to be installed the...")
- [ ] Review the page to ensure that there are no acronyms wrongly cased and all snippets are properly wrapped in their tags 

#### Dependencies
None

mbelluzzo, 2016-03-18:
"Add an introduction text to explain what these instructions are for" - OK
"briefly talk about what are the steps going to be taken", that is like:
int i = 6; // initialize integer var with value '6'

They know what is for, by reading the previously missing introduction, and read the page to know the steps, isn't that hard.

mbelluzzo, 2016-04-04:
"Review/reword Yocto instructions" -> Shouldn't this part be in meta-soletta wiki/instructions or meta-soletta tutorials and this page just point there? Is kind lost those instructions there (actually meta-soletta itself is pointing here). It should be like galileo instructions that teaches you how to recompile if you are using a image with meta-soletta, for dev purpose. Not to teach how to generate a full image.

gpaes, 2016-04-05:
@mbelluzzo 
"briefly talk about what the steps to be taken": this is not for each step but for the whole group of steps 

For the Yocto instructions I agree with you but the problem is that there's no meta-soletta wiki. If this content is useful for other pages and usages we could create a new page on the Soletta wiki itself.

mbelluzzo, 2016-04-05:
Even without a meta-soletta wiki, I think we should organize our wiki in a way that meta-soletta content is all group and controlled by its on pages (even that is on Soletta wiki) and we point there, so yes, I think we should create new pages within Soletta wiki for meta-soletta purpose and remove this info from this page. Will also make it possible to give descent instructions about minnowboard from Soletta context.

gpaes, 2016-04-06:
@mbelluzzo I canceled the yocto task, let's move forward with the meta-soletta page. I'ill open the tickets accordingly. Thanks for the input!!

closed 2016-04-06 by mbelluzzo