solettaproject/soletta #1655: Soletta Machine Learning: Light Sensor Tutorial: Improve page text

gpaes, 2016-03-18:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/Soletta-Machine-Learning:-Light-Sensor-Tutorial
On the [Overview](https://github.com/solettaproject/soletta/wiki/Soletta-Machine-Learning:-Light-Sensor-Tutorial#overview) section it'd be nice to talk about the problem that is going to be solved -- this is currently done in the [Coding](https://github.com/solettaproject/soletta/wiki/Soletta-Machine-Learning:-Light-Sensor-Tutorial#coding) section.

- [ ] Move the high level overview of the tutorial from Coding section to the Overview
- [ ] On Coding talk briefly about the steps that are going to be taking -- right now they are only listed as a bullet list

#### Dependencies
None

nirmalsinghania2008, 2016-03-19:
Updated!

bdilly, 2016-07-05:
thanks @nirmalsinghania2008

closed 2016-07-05 by bdilly

gpaes, 2016-07-11:
@nirmalsinghania2008 pretty nice contribution!:)