solettaproject/soletta #1673: API Changes: Sol-util

cabelitos, 2016-03-18:
#### Task Description
- [ ] Rename `SOL_UTIL_ARRAY_SIZE()`  to `sol_util_array_size()`
- [ ] Rename `sol_uint16_bytes_swap()` to `sol_util_uint16_bytes_swap()`
- [ ] Rename `sol_uint32_bytes_swap()` to `sol_util_uint32_bytes_swap()`
- [ ] Rename `sol_uint64_bytes_swap()` to `sol_util_uint64_bytes_swap()`
- [ ] Rename `SOL_NSEC_PER_SEC`  to `SOL_UTIL_NSEC_PER_SEC`
- [ ] Rename `SOL_MSEC_PER_SEC`  to `SOL_UTIL_MSEC_PER_SEC`
- [ ] Rename `SOL_USEC_PER_SEC`  to `SOL_UTIL_USEC_PER_SEC`
- [ ] Rename `SOL_NSEC_PER_MSEC`  to `SOL_UTIL_NSEC_PER_MSEC`
- [ ] Rename `SOL_NSEC_PER_USEC` to `SOL_UTIL_NSEC_PER_USEC`
- [ ] Rename `sol_util_secure_clear_memory()` to `sol_util_clear_memory_secure()`
- [ ] Rename `sol_util_strtodn()` to `sol_util_strtod_n()`
- [ ] Rename `sol_util_strtol()` to `sol_util_strtol_n()`
- [ ] Rename `sol_util_strtoul()` to `sol_util_strtoul_n()`
- [ ] Rename `sol_util_timespec_sum()` to `sol_util_timespec_add()`
- [ ] Rename `sol_util_uuid_str_valid()` to `sol_util_uuid_str_valid()`

#### Dependencies
None.

kaspersky, 2016-03-23:
@cabelitos , can you detail a bit the last rename?

cabelitos, 2016-03-23:
@kaspersky Sorry, it was a typo. Fixed now

closed 2016-05-11 by bdilly