solettaproject/soletta #1676: API Changes: Power Supply

bdilly, 2016-03-18:
#### Task Description

- [ ] Rename `sol_power_supply_exist()` to `sol_power_supply_exists()`
- [ ] Rename `sol_power_supply_get_voltage_max()` functions to `sol_power_supply_get_max_voltage()`
- [ ] Rename `sol_power_supply_get_voltage_min_design()` functions to `sol_power_supply_get_min_design_voltage()`
... => the same applies for all variants

nirmalsinghania2008, 2016-03-19:
@bdilly  what about sol_power_supply_get_voltage_max_design()?
should be change it to sol_power_supply_get_max_voltage_design()?

nirmalsinghania2008, 2016-03-19:
changed sol_power_supply_get_voltage_max(),sol_power_supply_get_voltage_min(),sol_power_supply_get_voltage_min_design(),sol_power_supply_get_voltage_max_design()

some for it you didn't mentioned .is it right?

closed 2016-06-13 by bdilly