solettaproject/soletta #1677: API Changes: Flow

bdilly, 2016-03-18:
#### Task Description

 - [ ] use unit8_t instead of unsigned char for byte representation
 - [ ] send_location_packet miss function receiving components, just like rgb and direction_vector does
 - [ ] rename `sol_flow_builder_export_in_port()` to `sol_flow_builder_export_port_in()` (the same applies for export_out_port
 - [ ] rename `sol_flow_packet_get_packet_type_as_string()` to `sol_flow_get_packet_type_name()`
 - [ ] function read_file() on struct sol_flow_parser_client should use sol_buffer
 - [ ] `sol_flow_parse_buffer_metatype()` should use sol_buffer too
 - [ ] `sol_flow_simplectype_event` struct should be renamed to `sol_flow_simple_c_type_event`
 - [ ] `sol_flow_simplectype_*` should be `sol_flow_simple_c_type` in the whole API
 - [ ] `sol_flow_single_port_in_connect()` should be `sol_flow_single_connect_port_in()` (same applies for disconnect and the remaining of API).

nirmalsinghania2008, 2016-03-19:
is it uint8_t or unit8_t?
unit8_t is included in stdint.h
i am working on it assuming it is uint8_t

kaspersky, 2016-05-19:
#2061

kaspersky, 2016-06-10:
#2155

ceolin, 2016-06-17:
done

closed 2016-06-17 by ceolin