solettaproject/soletta #1688: API Changes: LWM2M

cabelitos, 2016-03-21:
#### Task Description

- [ ] Do not use typedefs.
- [ ] Document what's a TLV 
- [ ] Rename `sol_lwm2m_client_info_get_sms()` to `sol_lwm2m_client_info_get_sms_number()`
- [ ] Rename `sol_lwm2m_notify_observers()` to `sol_lwm2m_client_notify()`
- [ ] Rename `sol_lwm2m_send_update()` to `sol_lwm2m_client_send_update()`
- [ ] `sol_lwm2m_tlv_get_bytes()` should receive a `struct sol_buffer` as argument
- [ ] Rename `sol_lwm2m_tlv_array_clear()` to `sol_lwm2m_tlv_list_clear()`
- [ ] Rename `sol_lwm2m_tlv_to_bool()`, `sol_lwm2m_tlv_to_int()`... and so on to
`sol_lwm2m_tlv_get_bool()`, `sol_lwm2m_tlv_get_int()` ...
- [ ] Rename `sol_lwm2m_server_management_write()` to `sol_lwm2m_server_write()`
- [ ] Rename `sol_lwm2m_server_management_read()` to `sol_lwm2m_server_read()`
- [ ] Rename `sol_lwm2m_server_management_delete` to `sol_lwm2m_server_delete_object_instance()`
- [ ] Rename `sol_lwm2m_server_management_execute()` to `sol_lwm2m_server_execute_resource()`
- [ ] Rename `sol_lwm2m_server_management_create` to `sol_lwm2m_server_create_object_instance()

#### Dependencies
None.

bsmelo, 2016-04-05:
I believe this is closed, isn't it? :-)

cabelitos, 2016-04-07:
@bsmelo Indeed. Closing

closed 2016-04-07 by cabelitos