solettaproject/soletta #1704: Strings on Soletta flows: Sum the node list up

gpaes, 2016-03-22:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/Strings-on-Soletta-flows
At the top section there's a big list of string nodes: replace it with most used ones and add a link to the Cheat Sheet for its full list.

#### Dependencies
None

bdilly, 2016-03-30:
@gpaes , while reading this page I saw that actually it had faulty links, but the explicit list of nodes has an important role on this page, if you click on each node type it jump to further explanation / usage example. So I believe we can keep them with fixed links.

Ptal and see if you agree it's good enough.

gpaes, 2016-03-30:
@bdilly Thanks for fixing the links! Let's leave them as is then. If needed we can revisit this.

closed 2016-03-30 by gpaes