solettaproject/soletta #1713: API Changes: Flow-Metatypes

cabelitos, 2016-03-23:
#### Task Description
- [ ] Remove typdefs
- [ ] Replace all occurences of metatype to meta_type. Example: `sol_flow_metatype_context` to `sol_flow_meta_type_context`
- [ ] Function `read_file` in `sol_flow_metatype_context` should receive a `struct sol_buffer` as argument instead of `const char **buf, size_t *size`

#### Dependencies

None.

barbieri, 2016-03-23:
I'm not sure i like meta_type more than metatype... :-D

bdilly, 2016-06-09:
ok, so let it as metatype

ceolin, 2016-06-14:
well, use typedef there will turn the illegible, i.e
`int (*(sol_flow_metatype_get_generate_code_start_func)(const struct sol_str_slice name))(const struct sol_flow_metatype_context *ctx, struct sol_buffer *out);`

i think is better let as it is.

ceolin, 2016-06-17:
done

closed 2016-06-17 by ceolin