solettaproject/soletta #1714: API Changes: Sol-Log

cabelitos, 2016-03-23:
#### Task Description
- [ ] Rename `SOL_LOG_LEVEL_POSSIBLE` to `SOL_LOG_LEVEL_IS_POSSIBLE`
- [ ] Rename `sol_log_get_show_colors()` to `sol_log_is_colors_enabled()`
- [ ] Rename `sol_log_get_show_file()` to `sol_log_is_file_name_enabled()`
- [ ] Rename `sol_log_get_show_function` to `sol_log_is_function_name_enabled()`
- [ ] Rename `sol_log_get_show_line` to `sol_log_is_line_number_enabled()`
- [ ] Rename `sol_log_set_show_colors()` to `sol_log_set_colors_enabled()`
- [ ] Rename `sol_log_set_show_file()` to `sol_log_set_file_name_enabled()`
- [ ] Rename `sol_log_set_show_function()` to `sol_log_set_function_name_enabled()`
- [ ] Rename `sol_log_set_show_line()` to `sol_log_set_line_number_enabled()`

#### Dependencies
None

barbieri, 2016-03-23:
are we sure about these "get" to "is"? Given that we have a "set" counterpart (ie: property), keeping get/set pairs is better, will make bindings expose those as properties.

@bdilly @glima @ibriano ?

ibriano, 2016-03-23:
On Wed, 23 Mar 2016 13:36:26 -0700, Gustavo Sverzut Barbieri wrote:
> are we sure about these "get" to "is"? Given that we have a "set" counterpart (ie: property), keeping get/set pairs is better, will make bindings expose those as properties.
> 
> @bdilly @glima @ibriano ?

I admit I haven't been paying attention to these renames, but I do like
_get_bla a lot better than _is_bla_enabled. The latter is too verbose for
me.

edersondisouza, 2016-03-23:
> counterpart (ie: property), keeping get/set pairs is better, will make
> bindings expose those as properties.
>
> I admit I haven't been paying attention to these renames, but I do like
> _get_bla a lot [...]

afair, in this case, the 'show' part on function names made it 
confusing, that's why the 'enabled' stuff was suggested.

barbieri, 2016-03-23:
then instead of enabled use displayed?

ibriano, 2016-03-23:
On Wed, 23 Mar 2016 14:33:47 -0700, Gustavo Sverzut Barbieri wrote:
> then instead of enabled use displayed?

sol_log_pardon_me_sir_but_i_would_like_to_ask_if_you_desire_the_filename_to_be_displayed()

amandawrcoelho, 2016-06-10:
Hey, what was decided here? @barbieri  @bdilly @ibriano

ibriano, 2016-06-10:
On Fri, 10 Jun 2016 13:05:49 -0700, Amanda Coelho wrote:
> Hey, what was decided here? @barbieri  @bdilly @ibriano 

No idea, I prefer the existing names.

bdilly, 2016-06-10:
For me initial naming is good, not confusing...
I agree with @ibriano , no me gusta big functions names =)

bdilly, 2016-06-10:
So closing it...

closed 2016-06-10 by bdilly

edersondisouza, 2016-06-10:
Well... it confused _everyone_ on review meeting...