solettaproject/soletta #1720: API Changes: COAP

cabelitos, 2016-03-23:
#### Task Description
- [ ] Remove typedefs
- [ ] Replace all `CONTENTTYPE`s with `CONTENT_TYPE`. Example `SOL_COAP_CONTENTTYPE_NONE` should be `SOL_COAP_CONTENT_TYPE_NONE`
- [ ] `LINKFORMAT` -> `LINK_FORMAT`
- [ ] `MAKE_RSPCODE` -> `sol_coap_make_response_code()`
- [ ] Replace `RSPCODE` with `RESPONSE_CODE`
- [ ] `responsecode` -> `response_code`
- [ ] `SOL_COAP_TYPE_NONCON` -> `SOL_COAP_TYPE_NON_CON`
- [ ] Use `size_t` for sizes instead of `uint16_t` - Examples - `sol_coap_add_option()` argument `len` should be a size_t
- [ ] `sol_coap_header_set_ver()` -> `sol_coap_header_set_version()`
- [ ] `sol_coap_header_get_ver()` -> `sol_coap_header_get_version()`
- [ ] `sol_coap_packet_notification_new()` -> `sol_coap_packet_new_notification()`
- [ ] `sol_coap_packet_request_new()` -> `sol_coap_packet_new_request()`
- [ ] `sol_coap_unobserve_server()` -> `sol_coap_unobserve_by_token()`
- [ ] `sol_coap_uri_path_to_buf()` -> `sol_coap_path_to_buffer()`
- [ ] Remove `sol_coap_secure_server_new()` and `sol_coap_server_new()` should receive a `bool` argument that flags if the server is secure or not

#### Dependencies
None.

otaviobp, 2016-05-13:
most of tasks fixed in #2024 
Assigning to @ceolin because he is working in last issue

cabelitos, 2016-06-20:
We're leaving the sizes as `uint16_t`
The last issue (`sol_coap_secure_server`) is fixed in https://github.com/solettaproject/soletta/pull/2185

cabelitos, 2016-06-20:
merged

closed 2016-06-20 by cabelitos