solettaproject/soletta #1725: Packages: Rename ArchLinux package soletta-git to soletta-nightly

gpaes, 2016-03-23:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/Packages
ArchLinux currently has two packages, soletta for releases and soletta-git for bleeding edge versions but both are git packages so the second one is not very meaningful. Renaming it to soletta-daily, soletta-nightly or something like that would help.

#### Dependencies
None

bdilly, 2016-03-24:
I'm wondering if -git isn't a standard way to name such kind of packages on Arch.

So let's see what @anselmolsm say about it.

anselmolsm, 2016-03-24:
-git is a standard for this kind o packages on Arch Linux

gpaes, 2016-03-24:
Thanks @bdilly and @anselmolsm, I'm going to close the ticket then!

closed 2016-03-24 by gpaes