solettaproject/soletta #1751: Logic constructs on Soletta flows: Add syntax highlight for C snippets

gpaes, 2016-03-28:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/Logic-constructs-on-Soletta-flows
There are a couple of C snippets without proper syntax highlighting -- adding the correct programming language name on each snippet will solve that.

#### Dependencies
None

bdilly, 2016-03-29:
Problem is github doesn't has support for FBP syntax.
But if tested it with prolog and result is good =)

closed 2016-03-29 by bdilly

edersondisouza, 2016-03-29:
> Problem is github doesn't has support for FBP syntax.
> But if tested it with prolog and result is good =)

Prolog? I was using sh... is it better?

gpaes, 2016-03-29:
@bdilly Prolog indeed yields good results! If there are no great harms I guess we should include this information bit to our **Contributing to the wiki** page.:)