solettaproject/soletta #1761: API Changes: questions / review / study

bdilly, 2016-03-28:
#### Task Description

Here is a list of some possible issues raised up by the team.

Before changing any of these a deeper review is required:

 - [ ] On I2C (but maybe other I/O API) - close_raw => should it be renamed ? better described ? Is it relevant to be exposed?
 - [ ] Are all sol_types required to be basic types for flow purposes? Do we need location and http response? Or could they just be composed types
 - [ ] Review `sol_flow_node_options_del()` - document  - may rename to del_options(). Apparently `type` shouldn't be const
 - [ ] Review `sol_flow_node_options_new()`. Does it really need to be exposed?

glima, 2016-05-20:
@mbelluzzo: the i2c one looks legit to me -- do we really need to expose the ``close_raw()`` thing? What's a valid use case?

glima, 2016-05-20:
I kinda dislike the idea of having to declare composed packets just to use location nodes. Also, we have @barbieri's idea of using memdesc on packets... The HTTP response packet has even ``sol_vector``s on it, no way to be composed.

glima, 2016-05-20:
I don't see why  ``type`` should **not** be ``const``. Do I miss anything?

glima, 2016-05-20:
The ``_new`` vs. ``_del`` pair pattern is broad in usage throughout Soletta and I think it's fair (``del_options`` would ruin that). I don't see how ``sol_flow_node_options_new()`` would not be exposed, as well.

glima, 2016-05-20:
One thing that pisses me off on our API is **const** in front of arguments passed by **value**, as in ``struct sol_aio *sol_aio_open_raw(const int device, const int pin, const unsigned int precision)`` -- what's the point?

bdilly, 2016-05-23:
@glima If I remember correctly what was discussed in the meeting that lead to this ticket, the problem with sol_flow_node_options_del() was that we don't have a "sol_flow_node_options", you act on "sol_flow_node", and the action would be to delete options, so it would be "sol_flow_node_del_options()".

cabelitos, 2016-05-23:
@glima I don't know why using const for arguments passed by value. We should remove it.

bdilly, 2016-05-23:
@glima regarding "const int" I also don't get the point. But I wasn't able to find much usage on Soletta API. Only aio and pinmux.

bdilly, 2016-05-23:
@glima I believe the discussion about "const type" was related to the fact you're changing the type (deleting options vector) and some devs could expect it wouldn't be changed by the function since it's "const".

mbelluzzo, 2016-05-23:
The discussion about "const type" didn't even happened to be honest. We just decided to remove those for value passing whenever found.

mbelluzzo, 2016-05-23:
About "close_raw", it has nothing to do with muxing like open_raw. I have no idea why @zehortigoza picked this name (since has a different purpose than open_raw) or even why it's needed.

bdilly, 2016-05-23:
@mbelluzzo "type" is a pointer. Two different items (const + value and const type).

mbelluzzo, 2016-05-23:
const <something that is not a pointer> in a function signature ->  we decided to remove those. It was let in by mistake by either the writer and the reviewer.

Of course const <pointer> in the function signature is still valid and very important.

cabelitos, 2016-05-24:
I have a PR that remove sol_i2c_close_raw()
https://github.com/solettaproject/soletta/pull/2091

bdilly, 2016-06-02:
After some discussion we weren't able to get a consensus about sol_flow_node_options_del() name changes, so we'll keep it the way it is.

Since it was the last item missing, I'm closing this ticket

closed 2016-06-02 by bdilly