solettaproject/soletta #1763: Logic constructs on Soletta flows: Improve page text

gpaes, 2016-03-29:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/Logic-constructs-on-Soletta-flows
The text on the page keeps making reference to the original code but it's not 100% clear what code this is. Is this the sample code? The code on the example above the text? Make it clear.
The phrase "“When doing flow based programming, one is advised to keep more than trivial logic controls inside their nodes, so that assembling a flow is only a matter of routing data through the right nodes.” is too complex, replace it with a new one to convey its idea: simpler code should be written as a custom node in order to avoid unnecessarily complex flows

- [ ] Make it clear what the "original code" actually is referring to
- [ ] Rewrite and simplify the "When doing flow based programming, ..." phrase
- [ ] Ensure that examples are ordered by their complexity

#### Dependencies
None

bdilly, 2016-05-30:
I believe the snippets order makes sense, other two points are valid

bdilly, 2016-05-30:
done

closed 2016-05-30 by bdilly