solettaproject/soletta #1771: Quickstart: Review terminal examples

gpaes, 2016-03-30:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/Quickstart
Review the terminal and sh examples to add syntax highlight and command prompt.

- [ ] Add syntax highlight to sh examples
- [ ] Add command prompt to terminal examples

#### Dependencies
None

bdilly, 2016-04-11:
Hey @gpaes , I wasn't able to find the sh examples you're mentioning.
Could you please be point me to it?

gpaes, 2016-04-11:
@bdilly for instance on the block:
$ export LD_LIBRARY_PATH=/path/to/lib/:$LD_LIBRARY_PATH
$ export PKG_CONFIG_PATH=/path/to/lib/pkgconfig:$PKG_CONFIG_PATH

Or do you think it's not useful to have them syntax-highlighted?

bdilly, 2016-05-11:
syntax-highlight is good, for sure

For sh on github it's not as interesting as it could be though (almost no changes at all)

Anyway, change done

closed 2016-05-11 by bdilly