solettaproject/soletta #1779: readme.md: Add direct download links to releases

gpaes, 2016-03-30:
#### Task Description
To make it easier to people to try Soletta out we must update readme.md with direct download to its stable releases for the supported boards. This section should be highlighted, being placed as high as possible on the document.

#### Dependencies
None

bdilly, 2016-04-01:
@gpaes  , not sure about this one.

Releases for supported boards may be done in different ways, we could have images generated with Ostro, Clear, regular Linux distros as Fedora or Arch.

So keeping all this data updated on README doesn't seem the best approach to me.

Couldn't we just keep it in our Home wiki page or something like that?

bdilly, 2016-04-01:
actually I believe my idea is something like what was proposed here: #1785

gpaes, 2016-04-04:
@bdilly That sounds pretty nice!

bdilly, 2016-04-06:
cool, so I'm closing this one

thanks

closed 2016-04-06 by bdilly