solettaproject/soletta #1789: Remove core.light and use oic.r.light.brigthness in soletta oic samples

bdilly, 2016-03-31:
#### Task Description

Now that we have OIC data models no need to provide custom resource types for it.

bdilly, 2016-04-05:
hmmm... actually it seems like they don't provide something like our light model.
They only provide the brightness. Most probably if it's off it should provide brightness == 0 and that's all.

In this case I'm inclined to let our custom data model (maybe even as an example of how to create custom stuff). What do you guys think about it?

@otaviobp @ceolin ^

otaviobp, 2016-04-13:
I think that we don't have a good light model in the IoT Data Models repository. We have 2 resources for similar cases (oic.r.light.dimming and oic.r.light.brightness), but we don't have a light.switch.
One guy used the oic.r.switch.binary resource as a resource to turn the lights on/off in one example in IoTivitity dev day. It would be nice to have one resource that is a light that includes oic.r.light.brightness and oic.r.switch.binary.
As the main IoTivity samples are using a resource that is like the core.light we have in Soletta, I would keep it and close this task. Let's keep following updates in the repository to see if they add something new to it.

bdilly, 2016-04-19:
@otaviobp yeah, it sounds like a plan

Let's see if they provide something similar soon. Closing it.

closed 2016-04-19 by bdilly