solettaproject/soletta #1799: Avoid adding CFLAGS that break debugging when building in debug mode

gabrielschulhof, 2016-04-01:
@ibriano wrote in https://github.com/solettaproject/soletta/pull/1798#discussion_r58211724:
> > On Fri, 01 Apr 2016 07:07:28 -0700, Flávio Vinicius Alvares Ceolin wrote:
> > > @@ -282,7 +282,7 @@ COMMON_CFLAGS += -fPIC
> > >  endif
> > >  
> > >  ifeq (y,$(BUILD_TYPE_DEBUG))
> > > -COMMON_CFLAGS += -g -fno-lto
> > > +COMMON_CFLAGS += -g -fno-lto -O0
> > >  endif
> > 
> > o yeah, here is much better
> 
> I'd rather re-order the way we add stuff to COMMON_CFLAGS to make sure
that all the things that can break debugging are not there in the first
place.

bdilly, 2016-06-13:
@ibriano , do you believe it's fixed with commit 170e100ac99?

ibriano, 2016-06-13:
We set flags in a more sane order now, but if you pick a release build, it will enable LTO and add -O2 (from the FORTIFY_SOURCE flags). For debug builds, there won't be any LTO enabled, and any optimization flags in the build will only come from the user.
I tried to make it so that if the user adds -g to CFLAGS, we disable flags that are not debug friendly, but I couldn't come up with any sane way to do that that would work across different compilers/platforms.

bdilly, 2016-06-13:
Nice.

I believe the problem reported was for debug mode (not release mode) and we're fine regarding that.

Indeed it would be nice to support even -g on release modes, but I believe it's really optional, since it would mean something like "give me a release mode that is not a release mode" =)

closed 2016-06-13 by bdilly