solettaproject/soletta #1820: API Changes: Review Comms API looking for convergence

bdilly, 2016-04-05:
#### Task Description

It would be nice to review all communication API and try to make they all look the same.

They should be async, callbacks should pass data pointer as first parameter, etc

But also they should look similar. E.g:
 * if in a protocol you add elements one by one using _add(*element) and other add an array of elements you should go with the same for both - if possible
 * if you initialized some modules instances passing a config and in others set config with a specific call

Go through all headers on `src/lib/comms/include/` and be happy

cabelitos, 2016-06-20:
The http-client code is already safe. Ignore the first missing item. I'm only creating a PR that update the docs.

https://github.com/solettaproject/soletta/pull/2191

cabelitos, 2016-06-21:
fixed

closed 2016-06-21 by cabelitos