solettaproject/soletta #1830: Meta-Soletta: Create page text

gpaes, 2016-04-06:
#### Task Description
Page: https://github.com/solettaproject/soletta/wiki/Meta-Soletta
Complete introductory text explaining why we have the meta-soletta layer and what's it for.
Add instructions for Edison, Galileo gen1 and gen2 and Minnowboard MAX -- checking first if they are close enough to be stated as a unified explanation.

- [ ] Complete introductory text
- [ ] Add instructions for **Edison**, **Galileo gen1 and gen2** and **Minnowboard MAX**, unifying the instructions for all of them if possible
- [ ] Update **meta-soletta/readme.md** to point to this new page

#### Dependencies
None

brunobottazzini, 2016-04-08:
@gpaes,

Two questions here:

In a near future, our officially images will be from Ostro OS (An IoT Os that is based in Yocto and uses our meta-soletta). 

1- In that case, should we put the instructions to build only using Ostro ? What do you think @barbieri 
?

2- Their build instructions would be huge. It may confuse the user when reading it. What do you think creating different pages for each board and then pointing out a link to them ?

gpaes, 2016-04-08:
@brunobottazzini 
1 - If the only official way is via Ostro, then yes, we would need to update the documents to talk about it only
2 - Yes, linking would be enough. We already have a link to Edison+Yocto

gpaes, 2016-04-12:
@brunobottazzini @bdilly I updated the ticket to reflect the new information. Please verifiy if it's all correct.

brunobottazzini, 2016-04-28:
Closing the issue...

As we both have reviewed the documentation and pushed to the master.

Thanks!!

closed 2016-04-28 by brunobottazzini