solettaproject/soletta #1886: flow: allow per-node-type aliases

barbieri, 2016-04-14:
Before, for ease of use, we had things like `grove/lcd-string`. But that component is not grove-dependent and may exist in other kits and boards. Then the proper way is to add a module per-component, allowing it to be used even without grove. This is also required because some of the grove components depend on I2C/AIO/SPI... while those may not be available in all platforms, that would disable the whole module or leave some "dummy" node types as we're doing now.

However to keep the usability in place we should offer per-node-type aliases, that can be declared in the `.json` as `"aliases": ["name1/xx", "name2/xx"]`. We should install the aliases to a standard location such as `$PREFIX/share/soletta/flow/aliases/00-name.json`, then the directory is read in order (so 00-99 can give precedence when needed) and we populate the alias-db.

The alias db can be checked in https://github.com/solettaproject/soletta/blob/master/src/shared/sol-conffile.c#L952 (`_resolve_config()`) after the known types are checked.

Check with @glima and @mbelluzzo as they were initial users of this feature.

cabelitos, 2016-04-19:
I will create a patch for this feature.

glima, 2016-04-28:
Done (PR).

closed 2016-04-28 by glima