solettaproject/soletta #1902: Making soletta compile with -Wformat and -Wformat-security

ereshetova, 2016-04-19:
#### Issue Description
Currently soletta does not compile if security compilation flags for format string vulnerabilities are enabled. 

#### Steps to reproduce

 1. add  "-Wformat -Wformat-security"  to CFLAGS
 2. run make

barbieri, 2016-04-19:
@ereshetova I just tried it and used `make menuconfig`, accessed `Basic > Compiler Options` and added `-Wformat -Wformat-security` to `Arbitrary CFLAGS`. It did work.

Also tried:
```sh
git clean -xfd
make alldefconfig CFLAGS="-Wformat -Wformat-security"
make all  CFLAGS="-Wformat -Wformat-security"
make check  CFLAGS="-Wformat -Wformat-security"
```
and it succeed. Could you paste your error?

My machine is running GCC5.3.0 in ArchLinux 32bits running in VirtualBox.

It is worth to note that we have been running with address `-fsanitizer=undefined` in our debug builds, running coverity (public checks) and nothing was reported about formatting strings in a while.

bdilly, 2016-05-11:
I wasn't able to reproduce it either.
I'm using gcc 5.3.1

Anyway these flags are used by default, as you can see at data/jsons/dependencies.json

So even if you just do your regular make alldefconfig with no extra CFLAGS these should get int.

ereshetova, 2016-05-11:
I think this issue can be closed since I can confirm that it does compile with the flags. Sorry for the confusion: it was due to other compilation issue on CI

closed 2016-05-11 by ereshetova