solettaproject/soletta #1941: Build error with latest Quark SOC v1.2 tool chain

lblim, 2016-04-27:
#### Issue Description

Need to right tool chain to cross-compile Soletta image for Galileo platform or clearer instruction that whether Soletta must be built with uclibc. The instructions on the webpage https://github.com/solettaproject/soletta/wiki/Galileo-and-Galileo-Gen2-Instructions points to the older Quark SOC X1000 BSP (v1.0.1), please update to the latest BSP (v1.2). 

For the v1.0.1 the SDK tool chain (http://download.intel.com/support/processors/quark/sb/quark_bsp_buildandswuserguide_329687_006.pdf) by default is uclibc, but the latest BSP (http://www.intel.com/content/dam/www/public/us/en/documents/release-notes/quark-x1000-bsp-build-sw-rel-user-guide.pdf) can output two different tool chain, for SPI image will use uclibc but DEFAULT full image will use glibc. 

I tried to build with the latest toolchains (both uclibc and glibc) are hitting into the same error:

      CC   build/stage/modules/flow/gpio/gpio.o
      CC   build/stage/modules/flow/aio/aio.o
      CC   build/stage/modules/flow/converter/converter.o
      CC   build/stage/lib/common/sol-blob.o
cc1: warning: ./build/stage/include/: No such file or directory
In file included from <command-line>:0:0:
/home/phoongst/mywork/soletta/src/lib/common/sol-missing.h:67:1: error: static declaration of 'memrchr' follows non-static declaration
memrchr(const void *haystack, int c, size_t n)
^
In file included from /home/phoongst/mywork/soletta/src/lib/common/sol-missing.h:64:0,
                 from <command-line>:0:
/opt/iot-devkit/1.7.2/sysroots/i586-poky-linux/usr/include/string.h:76:14: note: previous declaration of 'memrchr' was here
extern void *memrchr (const void *__s, int __c, size_t __n)
              ^
make: *** [build/stage/lib/common/sol-blob.o] Error 1

-------

I then tried to build the older tool chain based on the instruction, unfortunately it failed when building the tool chain

Running: git init
Initialized empty Git repository in /home/phoongst/mywork/quark_sdk/1.0.1/meta-clanton_v1.0.1/poky/.git/
Running: git remote add origin git://git.yoctoproject.org/poky
Running: git fetch origin --depth=3 tags/dylan-9.0.2:tags/dylan-9.0.2
remote: Counting objects: 5108, done.
remote: Compressing objects: 100% (4591/4591), done.
remote: Total 5108 (delta 332), reused 2736 (delta 206)
Receiving objects: 100% (5108/5108), 9.78 MiB | 742.00 KiB/s, done.
Resolving deltas: 100% (332/332), done.
From git://git.yoctoproject.org/poky
 * [new tag]         dylan-9.0.2 -> dylan-9.0.2
remote: Counting objects: 4, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 4 (delta 0), reused 4 (delta 0)
Unpacking objects: 100% (4/4), done.
 * [new tag]         dylan-9.0.2.rc1 -> dylan-9.0.2.rc1
 * [new tag]         yocot-1.4.2 -> yocot-1.4.2
 * [new tag]         yocto-1.4.2 -> yocto-1.4.2
 * [new tag]         yocto-1.4.2.rc1 -> yocto-1.4.2.rc1
Running: git checkout -b clanton tags/dylan-9.0.2
Switched to a new branch 'clanton'
Running: git init
Initialized empty Git repository in /home/phoongst/mywork/quark_sdk/1.0.1/meta-clanton_v1.0.1/meta-intel/.git/
Running: git remote add origin git://git.yoctoproject.org/meta-intel
Running: git fetch origin --depth=3 tags/dylan-9.0.2:tags/dylan-9.0.2
remote: Counting objects: 430, done.
remote: Compressing objects: 100% (274/274), done.
remote: Total 430 (delta 90), reused 326 (delta 64)
Receiving objects: 100% (430/430), 1.47 MiB | 502.00 KiB/s, done.
Resolving deltas: 100% (90/90), done.
From git://git.yoctoproject.org/meta-intel
 * [new tag]         dylan-9.0.2 -> dylan-9.0.2
remote: Counting objects: 11, done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 11 (delta 0), reused 11 (delta 0)
Unpacking objects: 100% (11/11), done.
 * [new tag]         1.4        -> 1.4
 * [new tag]         1.4.1.rc1  -> 1.4.1.rc1
 * [new tag]         1.4.final  -> 1.4.final
 * [new tag]         dylan-9.0.0 -> dylan-9.0.0
 * [new tag]         dylan-9.0.1 -> dylan-9.0.1
 * [new tag]         dylan-9.0.1.rc1 -> dylan-9.0.1.rc1
 * [new tag]         dylan-9.0.2.rc1 -> dylan-9.0.2.rc1
 * [new tag]         yocot-1.4.2 -> yocot-1.4.2
 * [new tag]         yocto-1.4.1 -> yocto-1.4.1
 * [new tag]         yocto-1.4.2 -> yocto-1.4.2
 * [new tag]         yocto-1.4.2.rc1 -> yocto-1.4.2.rc1
Running: git checkout -b clanton tags/dylan-9.0.2
Switched to a new branch 'clanton'
Running: git clone git://github.com/openembedded/meta-oe meta-oe
Cloning into 'meta-oe'...
fatal: remote error: 
  Repository not found.
Traceback (most recent call last):
  File "setup/gitsetup.py", line 131, in <module>
    main()
  File "setup/gitsetup.py", line 121, in main
    run_command('git clone %s %s' % (url, workdir))
  File "setup/gitsetup.py", line 65, in run_command
    subprocess.check_call(command, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'git clone git://github.com/openembedded/meta-oe meta-oe' returned non-zero exit status 128

#### Steps to reproduce

 1. Download the Soletta source code with tag v1_beta19
 2. Build the tool chains by referring to the documentation
 3. Build soletta

#### Board information

Host OS and architecture: Ubuntu 14.04.4 LTS

Target OS and platform: Linux, Galileo Gen 2

#### Logs

 * Soletta version (top commit hash): v1_beta19 (commit 6a7bc3)
 * .config: http://pastebin.com/h66D9zPq 
 * config log: http://pastebin.com/q9hTxmSf
[strace.zip](https://github.com/solettaproject/soletta/files/237869/strace.zip)

 * make log (V = 1): http://pastebin.com/1Js18jd
 * make strace log (strace -ff make &> strace.log): strace.zip
 *

brunobottazzini, 2016-04-27:
Hello @lblim,

Clanton is an outdated build for Yocto, and we highly recommend the use of Ostro OS.  

Ostro OS is a Yocto-based operating system for IoT devices.

Try reading this working-progress documentation here and follow the steps:
https://github.com/solettaproject/soletta/wiki/Meta-Soletta

Feel free to contact me if you have any doubt.

mbelluzzo, 2016-04-27:
Hey @lblim ,
Could you run 'make distclean'  (or even 'git clean -dxf' if you dont have uncommitted code) before trying the instructions to see if the problem is solved?

If you check your config file you can see that CONFIG_CFLAGS is empty, each means that you problaby compiled Soletta for your host before trying to compile for Galileo. After the environment setup, CFLAGS have the SDK info, so you should see that on your .config but is problaby not re-generated. The suggested commands should force that.

If it works, plz let us know so we can enforce those steps in the instructions text.

lblim, 2016-05-03:
I am building the Soletta on its own and not with Ostro because we have not migrated to ostro OS, Ostro OS does not have some of the ingredients of our BSP. I plan to build Soletta for Apollo Lake while waiting for my team to integrate soletta into our BSP.

lblim, 2016-05-03:
Hi @mbelluzzo 

I tried with make distclean and git clean -dxf with a clean checkout of soletta, tried with both uclibc and glibc toolchains, both reported same error, but this time the error is different from the previous error. I tried with the v1_beta19 and master branch.

make log = http://pastebin.com/piAkGjVw
.config = http://pastebin.com/GrYVqKwX
config.log = http://pastebin.com/bQeznQrz
[strace.zip](https://github.com/solettaproject/soletta/files/246225/strace.zip)

barbieri, 2016-05-03:
The logs are to worry about, it's mixing with your host system, things like `$PATH`, `$PKG_CONFIG_PATH`, `$PKG_CONFIG_LIBDIR` must be all changed to restrict access to your SDK.

also, you're missing lots of dependencies that are useful for a proper soletta application, like libcurl for http-client, libudev for device enumeration, libsystemd to control services and system state, openssl...

Specifically about make log (http://pastebin.com/piAkGjVw) you're trying to build micro-linux platform (instead of systemd, since you lack libsystemd) and that results in a failure since your `linux/reboot.h` header lack the suspend value `RB_SW_SUSPEND`.

uCLib is a poor choice of library as it misses lots of utilities we use in our linux port, like the printf() implementation is horrible and all. When attempting to do minimal systems, use musl-libc instead.

Last but not least, when you say "Ostro does not have your BSP", that's just a matter of adding your layer and using it with a higher priority than Ostro's. Please use the new instructions at https://ostroproject.org/documentation/howtos/building-images.html#adding-a-custom-layer-in-ostro-os 

Cross compiling complex software with multiple dependencies such as soletta is harder than you expect :-)

lblim, 2016-05-04:
Hi @Barbieri,

Thanks for the info, there are a few things I would like to suggest for the documentation update. It seems that it is not encourage to cross compile the soletta image due to dependencies, please update the page https://github.com/solettaproject/soletta/wiki/Galileo-and-Galileo-Gen2-Instructions accordingly with the working workflow or remove teh cross compile option.

I did try the uclibc and glibc and both hit into same prb, so should be the system issue. So looks like system is a MUST have component. The instruction actually asked to use Linux micro option and also the instruction asked to unset the $PKG_CONFIG_PATH etc.

Besides that, the build requirements (https://github.com/solettaproject/soletta/wiki/Build-requirements) may need to be updated since the libcurl, libudev are required and not optional.

My team is working on integrating the meta-soletta into APL-I yocto BSP now. I shall wait for them to enable it since that I am unable to proceed with cross-compile option.

Enabling Ostro OS for APL-I is in your roadmap now, so my team will pick up from there and add on meta layers that are missing to support our customers.

barbieri, 2016-05-04:
Hi Lee Booi,

Indeed the instructions need to be improved, there are multiple tasks pending on that :-)

The libraries are optional, but recommended. You may drop them, but you must know what you're doing. Usually would drop those in order to save memory.

It's strange glibc misses the symbol in `linux/reboot.h` since that exists since kernel 2.6. Maybe you missed kernel headers in your libc?

lblim, 2016-05-05:
Hi Gustavo,

I found the problem already. The GCC is v4.9.1 and the glibc is v2.19, this version actually has the definition of RB_SW_SUSPEND. RB_SW_SUSPEND is defined in sys/reboot.h and it is only added after v2.16 (2012 May). However, the sysroots of the SDK somehow has the older glibc header files. Therefore is a mismatch on the glibc header files version. I am not sure why this discrepancy happen yet, I will talk to quark team on that.

commit 1a09dc565db1da971ade18d9be7d9ac82646d599
Author: Andreas Jaeger <aj@suse.de>
Date:   Wed May 23 20:16:00 2012 +0200

    Add more constants to Linux <sys/reboot.h>
    
    2012-05-23  Andreas Jaeger  <aj@suse.de>
            Maximilian Attems  <max@stro.at>
    
        * sysdeps/unix/sysv/linux/sys/reboot.h (RB_SW_SUSPEND,
        RB_KEXEC):
        New macros.

barbieri, 2016-05-05:
ok... this is the reason why Ostro exists, the previous SDK for Galileo and Edison were painful to use, weird choice of components and can't scale to add more components (the traditional YP SDK is just the compiler, while the new eSDK embeds bitbake/devtool, then allow to add more recipes)

lblim, 2016-05-06:
I got the workaround for the toolchain already

Option 1 - Use the build environment SDK (created using "bitbake meta-ide-support")

Option 2 - If an SDK installer script is required (to install an SDK to a different computer) then make this simple change to meta/lib/oe/manifest.py and create the installer script again: Move PKG_TYPE_MULTILIB from last to first in the INSTALL_ORDER.

Before
     INSTALL_ORDER = [
        PKG_TYPE_LANGUAGE,
        PKG_TYPE_MUST_INSTALL,
        PKG_TYPE_ATTEMPT_ONLY,
        PKG_TYPE_MULTILIB
    ]

After
     INSTALL_ORDER = [
        PKG_TYPE_MULTILIB,
        PKG_TYPE_LANGUAGE,
        PKG_TYPE_MUST_INSTALL,
        PKG_TYPE_ATTEMPT_ONLY
     ]

However, I hit into another issue as below:

      CC   build/stage/modules/flow/console/console.o
      CC   build/stage/modules/flow/byte/byte.o
      CC   build/stage/modules/flow/led-7seg/led-7seg.o
      LD   build/soletta_sysroot/usr/lib/libsoletta.so
./src/lib/comms/sol-oic-server.c: In function 'res_payload_do.isra.0':
./src/lib/comms/sol-oic-server.c:239:1: internal compiler error: in propagate_rhs_into_lhs, at tree-ssa-dom.c:2913
 res_payload_do(CborEncoder *encoder,
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: i586-poky-linux-gcc returned 1 exit status
/opt/quark_sdk/1.7.2/sysroots/i586-pokysdk-linux/usr/bin/i586-poky-linux/../../libexec/i586-poky-linux/gcc/i586-poky-linux/4.9.1/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [build/soletta_sysroot/usr/lib/libsoletta.so] Error 1

Any idea?

barbieri, 2016-05-06:
Maybe you can try to remove `-ftlo` usage. Not sure if we have a way to do it now, @dorileo do you know?

This GCC 4.9 is ancient, LTO was just seeing its first users... much improved in 5.x series.

dorileo, 2016-05-06:
please run make with V=1 and show us the output, ```-flto``` will only be used if building with ```-O[0-5]```.

dorileo, 2016-05-06:
actually the other way around, ```-flto``` will only be used if **not** building with ```-O[0-5]```

barbieri, 2016-05-06:
strange, usually LTO is used with optimizations... are you sure about that?

Anyway, try the DEBUG build, AFAIR it doesn't include the LTO

dorileo, 2016-05-06:
@barbieri you're right, misread in the second run :(

```ifneq (,$(shell echo $(COMMON_CFLAGS) | grep -e "-O[s12345\ ]"))```

lblim, 2016-05-10:
.config : http://pastebin.com/4pGmTspJ
config.log : http://pastebin.com/1rBsEw7T
make log : http://pastebin.com/Bs5Yk4Yd 
[strace.zip](https://github.com/solettaproject/soletta/files/256472/strace.zip)

lblim, 2016-05-10:
Another build error with latest master branch code instead of v1_beta19, failed at make menuconfig

•Soletta version (top commit hash): master (commit 6f93cf1042)
.cpnfig : NIL
config.log : http://pastebin.com/EMHFYZEh
make log : http://pastebin.com/ETXwvaTu
make strace log (strace -ff make &> strace.log): strace.zip 
make strace log (strace -ff make menuconfig &> strace.log): strace2.zip
[strace.zip](https://github.com/solettaproject/soletta/files/256490/strace.zip)
[strace2.zip](https://github.com/solettaproject/soletta/files/256491/strace2.zip)

mbelluzzo, 2016-05-10:
This happens  when you forget to unset pkg_config_path ? This is usually
the case with this error.

lblim, 2016-05-11:
Nope, I have unset the PKG_CONFIG_PATH. This is the script I used to build and reproduce the prb

cd mywork/soletta
git pull 
make distclean
git clean -dfx
git submodule init && git submodule update
source /opt/quark_sdk/1.7.2/environment-setup-i586-poky-linux 
unset PYTHONHOME
export TARGETCC=$CC
export TARGETAR=$AR
unset PKG_CONFIG_PATH
make menuconfig

barbieri, 2016-05-11:
likely @mbelluzzo wrote one thing to mean the other: If you **set** `PKG_CONFIG_PATH` **and** `PKG_CONFIG_LIBDIR` you can avoid the cross compilation to look at your host.

That is, if you unset (as he wrote), then you're likely using `/usr/lib/pkgconfig` while you'd be interested in `/opt/quark_sdk/1.7.2/usr/lib/pkgconfig` (or something like that), that would include dependencies for libcurl, icu, mosquitto, microhttpd and so on.

barbieri, 2016-05-11:
Also, looking at your pastebins. I'd recommend to rebase to last master as we're improving some bits, I se `converter/string-format.c` but that's now at `format/string-format.c`, the warning looks harmless there.

The warning in `sol-memmap-storage.c:178` is also strange, `fread(&old_value, entry->size, 1, file);` with `old_value` being `uint64_t` (8 bytes), then `entry->size` is checked to be less or equal 8 with an assert: `assert(entry->size <= 8);`, so the supposed error from the compiler can never be reached.

The actual build error is resulting from an internal compiler error, that is, GCC itself is crashing.

My recommendation is to skip the old GCC and try a newer toolchain with 5.3, like the one in Ostro.

mbelluzzo, 2016-05-11:
Nope, I meant unset. We **unset** PKG_CONFIG_PATH so our host tools find the needed dependencies just fine (IOW do not look on the sdk to compile host stuff), like ncurses to display the menuconfig. And this doesn't break the cross compile because it's done through --sysroot flags everywhere and therefore should use the dir pointed at PKG_CONFIG_SYSROOT_DIR for everything related to it (iow the target dependencies), avoiding the mix of host and target dependencies. Unless something changed on yocto 1.7.2 environment script.

barbieri, 2016-05-11:
then you're lucky, because it makes no sense :-D `PKG_CONFIG_SYSROOT_DIR` affects `${prefix}` replacement.

Or pkg-config that is being used is the one provided by the SDK, usually patched to assume `PKG_CONFIG_LIBDIR` to the sdk installation path

mbelluzzo, 2016-05-11:
make V=1 while I cross compile says I'm not having luck, just pointing
things to the right stuff.

barbieri, 2016-05-11:
nice way to try this is to install a `-dev` package on your host that you do not have on your target, like `icu-dev` or something like this. See if `make reconf` is detecting it on your host and not on cross-compile.

mbelluzzo, 2016-05-11:
That is already the case when I compile Soletta to run on original Edison
(with the original SDK), and works just fine, including an observation that
"You're building the string nodes module without i18n support.", because
the dependency isn't on the target, but it is on my host.