solettaproject/soletta #1949: Failed to run example in the spec for GPIO on Ostro image#435

wanghongjuan, 2016-04-28:
#### Steps to reproduce

 1. Copy the example code to a file, named as gpio-example.js, then copy it to test device.
 2. Update the code:
```
var gpio = require("gpio");
```
change the require content "gpio" to the path of lib gpio.js and update
```
console.log("GPIO pin " + pin.pin " value changed to " + event.value);
```
to 
```
console.log("GPIO pin " + pin.pin " + value changed to " + event.value);
```
3. Run the js file
```
node gpio.js
```

#### Board information

Target OS and platform: Ostro image#435(https://download.ostroproject.org/builds/ostro-os/2016-04-26_13-41-09-build-435/) and Edison

#### Logs

 * Soletta version (top commit hash): tag(v1_beta19), commit id(6a7bc32)
Only show below message:
```
GPIO pin undefined open for writing
```
I think it should show all console log messages.
The other specifications(e.g. I2C, PWM, SPI,UART) have the similar situation as GPIO, the example in the specification can not access the test fluky.

baleboy, 2016-04-28:
According to @nagineni you should require the API with `require("soletta/gpio")`

wanghongjuan, 2016-04-28:
Unfortunately, I get the same issue. Is there need other precondition? for example, need to connect other sensors?

nagineni, 2016-04-28:
The example in the spec is incorrect, so please use the samples in this commit https://github.com/solettaproject/soletta/pull/1950/commits/06305496

wanghongjuan, 2016-04-28:
OK,  what about other specifications(e.g. I2C, PWM, SPI,UART)? they have the similar situation as GPIO, the example in the specification can not access the test fluky.

nagineni, 2016-04-28:
Look at samples in https://github.com/solettaproject/soletta/pull/1951/commits/73b31c2d2f for PWM and AIO. You should be able to run examples in the spec with little modifications for the other APIs. I will add more samples later.

wanghongjuan, 2016-04-28:
Thanks, I will try to run them.

zqzhang, 2016-04-29:
> The example in the spec is incorrect

@zolkis are you going to update the examples in the specs? Thanks.

commit 312358f2 (2016-05-12): JS spec: Fix require

This shall fix #1962 and close #1949.
closed 2016-05-12 by bdilly