solettaproject/soletta #1963: Failed to run upstream tests for Soletta Platform Services API on Ostro image#437

wanghongjuan, 2016-05-03:
#### Steps to reproduce
1.Run below command after clone the soletta repository.
```
$: cd soletta
$: npm install
```

2.Copy all relevant test files to test device (the folder “bindings” and some node modules that missing in test device(such as: “async”, “glob”, “nan”, “uuid”, “qunitjs” and “yargs”))
```
$: scp –r bindings root@[device-ip]:/usr/lib/node_modules/soletta/
$: scp –r node_modules/async root@[device-ip]:/usr/lib/node_modules/soletta/node_modules
```
Then copy other modules to device.

3.Enter the device and update the value of "prefix" to "invasive-tests" in file bindings/nodejs/tests/suite.js. 
```
    $: cd /usr/lib/node_modules/soletta/

     suiteOptions = _.extend( {

              // Default options
              prefix: "tests",
     }, require( "yargs" ).argv ),
```
4.Run command:
```
$: node bindings/nodejs/tests/suite.js
```

#### Board information

Target OS and platform: Ostro image#437(https://download.ostroproject.org/builds/ostro-os/2016-04-26_13-41-09-build-437/) and Edison
#### Logs
Soletta version (top commit hash): tag(v1_beta19), commit id(6a7bc32) Only show below message:
```
$: node bindings/nodejs/tests/suite.js
[Show following message]
$: root@edison:/usr/lib/node_modules/soletta# node bindings/nodejs/tests/suite.js
Main Loop Integration
NaN
```

barbieri, 2016-05-03:
I wonder why you're using "npm install" on soletta, you should use "make menuconfig", enable node.js, then "make install"

barbieri, 2016-05-03:
Also, the proper method is to use our Yocto recipe, the way you're doing may lead to problems since your host configuration may leak to target unless you fix many environment variables.

wanghongjuan, 2016-05-04:
@barbieri  The soletta project has been built in the Ostro image. The tests depend on some node modules what has been config in `package.json`.  The `npm` is variables in Ostro image . I tried to install the modules by command `npm install` on device and the issue was reproduced.

wanghongjuan, 2016-05-04:
The issue has been fixed by https://github.com/solettaproject/soletta/pull/1964.

closed 2016-05-05 by wanghongjuan