solettaproject/soletta #1968: Avoid depending on Duktape master branch

svaarala, 2016-05-03:
It seems Soletta is using Duktape master branch (git submodule) at the moment. Duktape master branch will soon get API incompatible changes as development Duktape 2.0 is starting, so it might be good to change the branch from master to a release tag (e.g. `v1.5.0`) or `v1-maintenance` for the latest 1.x version.

bdilly, 2016-05-04:
@svaarala thanks a lot for commenting about it.

Actually we're pointing to the master branch of a custom duktape repository:
https://github.com/solettaproject/duktape-release

We tried to just use https://github.com/svaarala/duktape-releases but actually what is provided are tarballs.

All the matters from Soletta POV are the generated files, duktape.[c|h]

And we would like to get it in a straight-forward to just use on Soletta, and avoid to push people to download a lot of data.

Don't you think it could be useful for other project to provide a repository like that?

Thanks in advance

svaarala, 2016-05-04:
Duktape-releases repository provides unpacked distributables as tags. For example, the 1.5.0 release is:

- https://github.com/svaarala/duktape-releases/tree/v1.5.0

svaarala, 2016-05-04:
In any case the issue in the title is happily a non-issue so this can be closed :) I could open a Duktape issue about providing a subset distributable if the tarballs or unpacked tarballs in duktape-releases are awkward to use in your case?

dorileo, 2016-05-04:
duktape-releases repository is a huge repository, with huge objects, this makes unfeasible to be used for example on semaphore, this is something we wouldn't like to deal with simply to get 3 files.

@svaarala could you manage to maintain any kind of duktape-thin-releases where we could grab only the required source code? or any sort of? :-)

Thanks...

svaarala, 2016-05-04:
Hmm, I could add a repository which only hosts the unpacked distributable packages (and not the tarballs). That wouldn't be a tiny repository either; duktape-releases is about 60MB and only 23MB of that is from the tarballs.

Using a subset of the distributable would also be possible - but which subset? The majority of the distributable consists of the combined and separate source files, and the rest (like the "genconfig" config file tool) is needed in some projects and not in others, so I'm not really sure what subset would be useful (and not lead to going back and forth on the subset :-). I'm open to suggestions :)

bdilly, 2016-05-05:
But other projects make use of the separate source files?
At first my idea was to just provide the combined duktape.c and header.

Given the project documentation (step 1 at http://duktape.org/) I understood it is the most frequent use case.

But certainly you're in a much better position to see if having a release repository of the combined files only could help more projects or not.

Maybe we have a very specific requirement on Soletta project.

barbieri, 2016-05-05:
indeed, for consumers having the "single source" version is much more convenient, so a repo with only that (autogenerated) and able for us to do "maxdepth=1" clones would be much nicer :-)

svaarala, 2016-05-05:
@bdilly I think most projects use the single source distributable (although there are actually two variants of that already, with and without `#line` directives) but I don't think people usually mind downloading the ~700kB distributable to get the source. Maintenance-wise it's of course easier the fewer different supported deliverables there are.

@barbieri How about cloning a release tag from duktape-releases with max depth 1?  For example:

    git clone https://github.com/svaarala/duktape-releases --depth 1 --branch v1.5.0 --single-branch duktape-1.5.0.,

creates an output directory which is more or less the same size as an unpacked distributable tarball (e.g. 14MB for 1.5.0 clone, vs. 12MB for an unpacked distributable tarball). The single source directory (src/ in the distributable) is 3.3MB so it's still larger of course than just getting the single source.

Anyway, opened an issue for tracking this: https://github.com/svaarala/duktape/issues/748.

barbieri, 2016-05-05:
For us the submodule is simpler, I recall @glima or @bdilly had a special need for repository format or branch so we could use submodule and max depth of 1.

glima, 2016-05-05:
We only require the no ``#line``s version of the distribution

bdilly, 2016-05-05:
@svaarala once again, thanks for bringing up this discussion.

I'll create a task here to move to the repo to be created as outcome of https://github.com/svaarala/duktape/issues/748 and I'll keep an eye on it.

As soon as it's done we'll drop our.

closed 2016-05-05 by bdilly