solettaproject/soletta #1979: Add support to disable modules that are not used/needed

jimmy-huang, 2016-05-04:
#### Task Description
Currently when building on Zephyr, to use Soletta to build a simple GPIO app, some of the libraries are stilll being built and linked although the app do not use these libraries or modules.  Some can be disabled through menuconfig like LOG, FLOW_SUPPORT, USE_UPDATE, but some can't be disabled like crypto, buffer, random, vector (or maybe they do but not well-documented or through menuconfig), It would be nice if they can be configured as well so the app can be build with the absolute minimal config that includes the mainloop, gpio and whatever neccesary to function in order to further reduce the library size.

#### Steps to reproduce

 1.  Current sol.conf
NETWORK=n
FLOW_SUPPORT=n
USE_UPDATE=n
LOG=n

 2. soletta modules gets built and linked with this config
lib/soletta/lib.a(sol-mainloop.o)
                              ../apps/soletta_test/src/built-in.o (sol_timeout_add)
lib/soletta/lib.a(sol-platform.o)
                              lib/soletta/lib.a(sol-mainloop.o) (sol_platform_init)
lib/soletta/lib.a(sol-mainloop-common.o)
                              lib/soletta/lib.a(sol-mainloop.o) (sol_mainloop_impl_init)
lib/soletta/lib.a(sol-mainloop-impl-zephyr.o)
                              lib/soletta/lib.a(sol-mainloop-common.o) (sol_mainloop_impl_platform_init)
lib/soletta/lib.a(sol-platform-impl-zephyr.o)
                              lib/soletta/lib.a(sol-platform.o) (sol_platform_impl_init)
lib/soletta/lib.a(sol-gpio-common.o)
                              ../apps/soletta_test/src/built-in.o (sol_gpio_open)
lib/soletta/lib.a(sol-gpio-impl-zephyr.o)
                              ../apps/soletta_test/src/built-in.o (sol_gpio_close)
lib/soletta/lib.a(sol-modules.o)
                              lib/soletta/lib.a(sol-mainloop.o) (sol_modules_clear_cache)
lib/soletta/lib.a(sol-monitors.o)
                              lib/soletta/lib.a(sol-platform.o) (sol_monitors_init_custom)
lib/soletta/lib.a(sol-util.o)
                              lib/soletta/lib.a(sol-platform.o) (sol_util_replace_str_if_changed)
lib/soletta/lib.a(sol-random.o)
                              lib/soletta/lib.a(sol-util.o) (sol_random_new)
lib/soletta/lib.a(sol-util-impl-zephyr.o)
                              lib/soletta/lib.a(sol-mainloop-common.o) (sol_util_timespec_get_current)
lib/soletta/lib.a(sol-buffer.o)
                              lib/soletta/lib.a(sol-util.o) (sol_buffer_expand)
lib/soletta/lib.a(sol-str-table.o)
                              lib/soletta/lib.a(sol-gpio-common.o) (sol_str_table_lookup_fallback)
lib/soletta/lib.a(sol-vector.o)
                              lib/soletta/lib.a(sol-monitors.o) (sol_vector_init)
lib/soletta/lib.a(sol-crypto.o)
                              lib/soletta/lib.a(sol-mainloop.o) (sol_crypto_init)
lib/soletta/lib.a(sol-blob.o)

#### Board information

Host OS and architecture:
Ubuntu 14.04 64bit

Target OS and platform:
Zephyr Nanokernel and ARduino 101

Soletta version (top commit hash): 3354c99fdf7b494e7aa664cefa42dd2570c53b55

bdilly, 2016-05-05:
Even if you're not using it directly eventually other modules that you need are using them.

Sure, we should review that and see if all of them are really required.
Maybe some of them could be avoided if not used, maybe with lazy loads / init

For example, only initialize crypto function when first use is required instead of initializing on mainloop?

@glima are you facing issues like that?

glima, 2016-05-05:
@jimmy-huang, bringing us just the .o's list is not fair. Please paste <...>/zephyr-sdk-0.7.5/sysroots/i686-pokysdk-linux/usr/bin/iamcu-poky-elfiamcu/i586-poky-elfiamcu-nm your-image.elf's output for us, so we can check that indeed we got all those you cited being used. In your config you did not turn off crypto, for example, so it's OK to see it being compiled, but after a "-Os" compiler flag it should go away.

glima, 2016-05-05:
Moreover, buffer and vector data types are used thoroughly inside Soletta, I'm afraid you can't live without those

barbieri, 2016-05-05:
Indeed, we use the datatypes ourselves. And although some modules are not optional, we do spill each data and function in their own sections (`-fdata-sections -ffunction-sections`), then what is not used will go away (`-Wl,--gc-sections`), then say you do not use the `sol_platform_*` family, their symbols will just go away automatically. Usually there will be one reference to the `_init()` and `_shutdown()`, but they are harmless as they are carefully doing as few stuff as possible. This is the case for crypto, platform and so on :-)

If you're looking to save space, focus on log, usually it's what adds lots of space. Be aware that turning off the log (LOG=n) will make your debug harder, so my recommendation is to use a maximum log level during development, so you keep critical/error/warning, remove info/debug. This should be reasonable.

glima, 2016-05-10:
@jimmy-huang, so anything new? If no evidence that the stated title is indeed an issue, we're going to close this, OK?

jimmy-huang, 2016-05-10:
@glima closing this as this is working as intended.

closed 2016-05-10 by jimmy-huang