solettaproject/soletta #1982: proposal: use typedef for public opaque structs

cmarcelo, 2016-05-05:
The page https://github.com/solettaproject/soletta/wiki/API-design-conventions documents Soletta approach for dealing with typedefs:

> We don't typedef structs, enums, function pointers. There are some exceptions around the code, but we prefer not to open space for more.

I propose to add an exception for pointers to opaque structures, at least when used by consumers of the library. For the consumer of the library, there is literally no difference whether `sol_gpio *` is a struct or not, because they can't access its contents.

Note the proposal is to typedef the structs not pointers to the structs. This doesn't necessarily affects how code inside Soletta is written.

The exception won't apply for structs that are not used only via opaque pointers: I do think we would benefit in this case as well, but it's a harder argument to have, and IMHO there is some value in this limited scope proposal.

cmarcelo, 2016-05-05:
@bdilly @glima @ibriano

bdilly, 2016-05-05:
@cmarcelo could you please share the reasons / pros on having these typedefs?
How does it help you?

cmarcelo, 2016-05-05:
It's literally typing less code. From the user perspective (different from when the struct is not opaque), there's zero benefit from knowing the handle I got is a struct. It makes variable declarations and function parameter lists longer.

The program below has 4 struct keywords that provide zero value for the user of Soletta.

```
static struct sol_gpio *button;
static struct sol_gpio *led_red;
static struct sol_gpio *led_green;

static void on_button(void *data, struct sol_gpio *gpio, bool value) {
	sol_gpio_write(led_red, value);
	sol_gpio_write(led_green, !value);
}
```

vs.

```
static sol_gpio *button;
static sol_gpio *led_red;
static sol_gpio *led_green;

static void on_button(void *data, sol_gpio *gpio, bool value) {
	sol_gpio_write(led_red, value);
	sol_gpio_write(led_green, !value);
}
```

For completeness, the usual argument *against* typedef I've found was the one from Linux kernel: http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/Documentation/CodingStyle#n273 in their style, I would even fit this case in (a), since the user won't be able to see or change the contents of the struct.

barbieri, 2016-05-05:
+1 for me

otaviobp, 2016-05-11:
+1 too

edersondisouza, 2016-05-11:
> It's literally typing less code. From the user perspective (different
> from when the struct is not opaque), there's zero benefit from knowing
> the handle I got is a struct. It makes variable [...]

Internally, on Soletta code, such typedef shouldn't be used? Is it  only 
for Soletta users?

cmarcelo, 2016-05-11:
> Internally, on Soletta code, such typedef shouldn't be used? Is it  only for Soletta users?

In the original proposal I run away from deciding this:

> This doesn't necessarily affects how code inside Soletta is written.

In the long run, I think will be easier to embrace typedef, but from feedback in IRC some people would be against it. Making they available for the users is still a win even if Soletta developers write the library code explicitly writing "struct".

cabelitos, 2016-06-21:
+1
Can we add a proposal to do de same thing for callbacks :new_moon_with_face:  ?

cmarcelo, 2016-07-18:
Finally wrote the patch for this (linked above).

The original proposal (and the patch) were very conservative: just adding for the opaque types, we end up adding another variable to the user "some sol_ has struct in front others not". For example when using GPIO we have both sol_gpio_config and sol_gpio, one would need the struct and the other wouldn't.

I thought this intermediate state would be OK, but now I think its a bad. I'm happy to upgrade that patch to add the extra typedefs for the other structs.

@bdilly ACK/NACK/Thoughts?

bdilly, 2016-07-18:
@cmarcelo I agree having it for all of them would be less confusing for users.

bdilly, 2016-07-18:
@cmarcelo also please feel free to do a regular pull request so it would be checked by semaphoreci, etc

commit 342b9113 (2016-07-19): all: add typedefs to public API structs

Soletta already uses "sol_" namespace and most of the time a secondary
namespace that is module-specific. This makes the identifiers very
large. By removing the need of explicitly use "struct" the client code
has less noise.

The convention for Doxygen documentation was:

- Opaque structs appear in typedefs section of the
  documentation. @typedef doxygen marker is used.

- Public structs appear both in data structures and typedefs
  sections. The documentation is not marked so it applies to both (some
  markers were removed).

This patch includes a cha [... cut]
closed 2016-07-19 by cmarcelo